frankarobotics/franka_ros #4: Mesh appearance

gavanderhoorn, 2017-10-08:
This is not a complaint, but just curiosity: are you planning to release meshes other than STLs for the non-collision geometry? The robot is completely white in RViz for me at the moment, which is almost an offense to the aesthetics of the real robot.

Colladas with perhaps some texturing would be a huge improvement in visual appearance :)

fwalch, 2017-10-09:
Thanks for the suggestion! We'll discuss this internally.

gavanderhoorn, 2018-01-25:
Just a friendly 🛎 : have you guys made any progress with this?

The white is just too bright ;)

fwalch, 2018-01-26:
Sorry it's taking so long, but I hope we'll soon have updated meshes in `franka_ros`. I'll leave a comment here when this has happened.

fwalch, 2018-02-23:
With the new 0.3.0 release, we now have Collada meshes in `franka_description` :)

closed 2018-04-16 by fwalch