frankarobotics/franka_ros #9: catkin_build

BluePackStack, 2018-01-31:
Seid kurzem funktioniert die Anleitung zur Installation von franka_ros für mich nicht mehr.

catkin_make -DCMAKE_BUILD_TYPE=Release -DFranka_DIR:PATH=/path/to/libfranka/build
oder catkin_make bewirken folgende Fehlermeldung.
```

-- +++ processing catkin package: 'franka_visualization'
-- ==> add_subdirectory(franka_ros/franka_visualization)
CMake Error at franka_ros/franka_visualization/CMakeLists.txt:12 (find_package):
  Could not find a configuration file for package "Franka" that is compatible
  with requested version "0.2.0".

  The following configuration files were considered but not accepted:

    /home/doom/libfranka/build/FrankaConfig.cmake, version: 0.1.1
    /opt/ros/kinetic/lib/x86_64-linux-gnu/cmake/Franka/FrankaConfig.cmake, version: 0.1.0

-- Configuring incomplete, errors occurred!
See also "/home/doom/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/doom/catkin_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```

ch1bo, 2018-02-01:
Das aktuelle franka_ros release benötigt libfranka-0.2.0. Sie müssen also zuerst libfranka updaten und installieren. Bei Ihnen scheint dies bei /home/doom/libfranka zu sein.

⁣Sent from TypeApp ​

BluePackStack, 2018-02-02:
Besten Dank! Hat geklappt.

closed 2018-02-02 by BluePackStack