frankarobotics/franka_ros #10: catkin_make Franka Ros

vitomillan, 2018-02-01:
Beim kompilieren von Franka-Ros 

`~/catkin_ws$ catkin_make -DCMAKE_BUILD_TYPE=Release -DFranka_DIR:PATH=~/libfranka/build`

tritt folgender Fehler auf:

> In function `main':
franka_joint_state_publisher.cpp:(.text.startup+0x5a0): undefined reference to `franka::Robot::Robot(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, franka::RealtimeConfig)'
collect2: error: ld returned 1 exit status

`
Base path: /home/mti/catkin_ws
Source space: /home/mti/catkin_ws/src
Build space: /home/mti/catkin_ws/build
Devel space: /home/mti/catkin_ws/devel
Install space: /home/mti/catkin_ws/install
####
#### Running command: "cmake /home/mti/catkin_ws/src -DCMAKE_BUILD_TYPE=Release -DFranka_DIR:PATH=~/libfranka/build -DCATKIN_DEVEL_PREFIX=/home/mti/catkin_ws/devel -DCMAKE_INSTALL_PREFIX=/home/mti/catkin_ws/install -G Unix Makefiles" in "/home/mti/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/mti/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /opt/ros/kinetic
-- This workspace overlays: /opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/mti/catkin_ws/build/test_results
-- Found gtest sources under '/usr/src/gtest': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.8
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 9 packages in topological order:
-- ~~  - franka_description
-- ~~  - franka_ros (metapackage)
-- ~~  - franka_msgs
-- ~~  - franka_visualization
-- ~~  - franka_gripper
-- ~~  - franka_hw
-- ~~  - franka_control
-- ~~  - franka_example_controllers
-- ~~  - panda_moveit_config
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- +++ processing catkin package: 'franka_description'
-- ==> add_subdirectory(franka_ros-kinetic-devel/franka_description)
-- +++ processing catkin metapackage: 'franka_ros'
-- ==> add_subdirectory(franka_ros-kinetic-devel/franka_ros)
-- +++ processing catkin package: 'franka_msgs'
-- ==> add_subdirectory(franka_ros-kinetic-devel/franka_msgs)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- franka_msgs: 2 messages, 0 services
-- +++ processing catkin package: 'franka_visualization'
-- ==> add_subdirectory(franka_ros-kinetic-devel/franka_visualization)
-- +++ processing catkin package: 'franka_gripper'
-- ==> add_subdirectory(franka_ros-kinetic-devel/franka_gripper)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Generating .msg files for action franka_gripper/Grasp /home/mti/catkin_ws/src/franka_ros-kinetic-devel/franka_gripper/action/Grasp.action
-- Generating .msg files for action franka_gripper/Homing /home/mti/catkin_ws/src/franka_ros-kinetic-devel/franka_gripper/action/Homing.action
-- Generating .msg files for action franka_gripper/Stop /home/mti/catkin_ws/src/franka_ros-kinetic-devel/franka_gripper/action/Stop.action
-- Generating .msg files for action franka_gripper/Move /home/mti/catkin_ws/src/franka_ros-kinetic-devel/franka_gripper/action/Move.action
-- franka_gripper: 28 messages, 0 services
-- +++ processing catkin package: 'franka_hw'
-- ==> add_subdirectory(franka_ros-kinetic-devel/franka_hw)
-- +++ processing catkin package: 'franka_control'
-- ==> add_subdirectory(franka_ros-kinetic-devel/franka_control)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Generating .msg files for action franka_control/ErrorRecovery /home/mti/catkin_ws/src/franka_ros-kinetic-devel/franka_control/action/ErrorRecovery.action
-- franka_control: 7 messages, 7 services
-- +++ processing catkin package: 'franka_example_controllers'
[... 181 lines cut ...]
[ 88%] Built target franka_example_controllers_generate_messages_nodejs
Scanning dependencies of target franka_example_controllers_generate_messages_py
[ 88%] Generating Python from MSG franka_example_controllers/JointTorqueComparison
[ 89%] Linking CXX executable /home/mti/catkin_ws/devel/lib/franka_visualization/franka_joint_state_publisher
[ 89%] Built target franka_example_controllers_generate_messages_cpp
Scanning dependencies of target franka_example_controllers_generate_messages_eus
[ 90%] Generating EusLisp code from franka_example_controllers/JointTorqueComparison.msg
[ 90%] Generating EusLisp manifest code for franka_example_controllers
[ 91%] Generating Python msg __init__.py for franka_example_controllers
CMakeFiles/franka_joint_state_publisher.dir/src/franka_joint_state_publisher.cpp.o: In function `main':
franka_joint_state_publisher.cpp:(.text.startup+0x5a0): undefined reference to `franka::Robot::Robot(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, franka::RealtimeConfig)'
collect2: error: ld returned 1 exit status
franka_ros-kinetic-devel/franka_visualization/CMakeFiles/franka_joint_state_publisher.dir/build.make:114: recipe for target '/home/mti/catkin_ws/devel/lib/franka_visualization/franka_joint_state_publisher' failed
make[2]: *** [/home/mti/catkin_ws/devel/lib/franka_visualization/franka_joint_state_publisher] Error 1
CMakeFiles/Makefile2:1166: recipe for target 'franka_ros-kinetic-devel/franka_visualization/CMakeFiles/franka_joint_state_publisher.dir/all' failed
make[1]: *** [franka_ros-kinetic-devel/franka_visualization/CMakeFiles/franka_joint_state_publisher.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 91%] Built target franka_example_controllers_generate_messages_py
[ 91%] Built target franka_example_controllers_generate_messages_eus
[ 92%] Linking CXX shared library /home/mti/catkin_ws/devel/lib/libfranka_hw.so
[ 92%] Built target franka_hw
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
`

fwalch, 2018-02-02:
Dieser Fehler tritt auf, wenn eine alte Version von `ros-kinetic-libfranka` installiert ist. Nach der Deinstallation (`sudo apt remove ros-kinetic-libfranka`) sollte das Kompilieren von `franka_ros` funktionieren. Wir werden aber auch versuchen, das zugrundeliegende Problem zu beheben, sodass es nicht mehr zu solchen Konflikten kommt.

vitomillan, 2018-02-02:
Es scheint leider eine andere Ursache zu haben. Libfranka und ros-kinetic-libfranka und der panda-server sollten bereits die aktuellste Version haben.

`
catkin_make -DCMAKE_BUILD_TYPE=Release -DFranka_DIR:PATH=/libfranka/build -j 1`

liefert ebenfalls folgenden Fehler:

`[  5%] Linking CXX executable /home/mti/catkin_ws/devel/lib/franka_visualization/franka_joint_state_publisher
CMakeFiles/franka_joint_state_publisher.dir/src/franka_joint_state_publisher.cpp.o: In Funktion `main':
franka_joint_state_publisher.cpp:(.text.startup+0x5a0): Nicht definierter Verweis auf `franka::Robot::Robot(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, franka::RealtimeConfig)'
collect2: error: ld returned 1 exit status
franka_ros/franka_visualization/CMakeFiles/franka_joint_state_publisher.dir/build.make:114: die Regel für Ziel „/home/mti/catkin_ws/devel/lib/franka_visualization/franka_joint_state_publisher“ scheiterte
make[2]: *** [/home/mti/catkin_ws/devel/lib/franka_visualization/franka_joint_state_publisher] Fehler 1
CMakeFiles/Makefile2:1166: die Regel für Ziel „franka_ros/franka_visualization/CMakeFiles/franka_joint_state_publisher.dir/all“ scheiterte
make[1]: *** [franka_ros/franka_visualization/CMakeFiles/franka_joint_state_publisher.dir/all] Fehler 2
Makefile:138: die Regel für Ziel „all“ scheiterte
make: *** [all] Fehler 2
Invoking "make -j 1" failed
`

fwalch, 2018-02-02:
>  Libfranka und ros-kinetic-libfranka und der panda-server sollten bereits die aktuellste Version haben.

Die aktuellste Version von `libfranka` ist noch gar nicht in den ROS-Repos vorhanden; die alte Version muss unbedingt deinstalliert werden, damit die Kompilierung funktioniert. Haben Sie `sudo apt remove ros-kinetic-libfranka` ausgeführt? Haben Sie vielleicht eine andere Version von `libfranka` installiert (`sudo apt remove libfranka`)?

vitomillan, 2018-02-02:
Danke für den Hinweis. Dass die libfranka in den ROS-Repos noch nicht aktuell ist, ist einem Kollegen dann später auch aufgefallen. Letztlich wurden alle Dateien der älteren Version manuell vom System entfernt und dann erfolgreich kompiliert.

Besten Dank!

closed 2018-02-02 by vitomillan