frankarobotics/franka_ros #12: feature request: inertial tags in panda xacro

tstoyanov, 2018-02-14:
Hi,
using the panda robot in gazebo with roscontrol requires that all links have inertial properties, which at the moment they do not. It would be great to add these parameters, as in my experience running a couple of preliminary tests in gazebo is really convenient. 

Thanks for the great work on this package :)

carlosjoserg, 2018-03-08:
Related to what would be within those tags...

From what I could see through the [`libfranka`](https://github.com/frankaemika/libfranka) repo and [documentation](https://frankaemika.github.io/docs/overview.html?highlight=model), it looks like the dynamic model is in the FCI/Control unit. Then, it is loaded via a [connection to the robot](https://github.com/frankaemika/libfranka/blob/master/examples/joint_impedance_control.cpp#L125), so one can query the dynamic parameters given a robot state via the [robot model library](https://github.com/frankaemika/libfranka/blob/master/examples/joint_impedance_control.cpp#L183), e.g. the coriolis term in the previous link. 

That'd mean that, if I don't have an actual robot, I can't read those parameters, right?
If so, is that because some sort of online parameter identification is running on the FCI/Control unit or just that the dynamic model can't be publicly available?

fwalch, 2018-10-19:
FYI, @davetcoleman's talk at ROSCon [showed how the MoveIt setup assistant 2.0 will be able to generate inertial tags for Gazebo integration](https://vimeo.com/293523535#t=983s).

sven-hoek, 2018-11-28:
Did anyone of you get to simulate the robot in Gazebo? I'm working on it right now and I'd prefer not to do work that has already been done. Or some help/exchange of knowledge is generally appreciated.

For now I've read into what Gazebo needs and started to insert dummy inertia values into the robot description. Will continue with the transmissions etc. and hope to get at least the standard interfaces/controllers and grasping running.

hu-yue, 2018-11-30:
I would like to add my two cents.
We were interested in using the Panda in Gazebo as well, however, after discovering that it is not possible to have a model with the actual proper dynamic parameters, we had to abandon the idea of using it.
We do have the actual robot, but we wanted to run proper simulations in Gazebo instead of directly running things on the robot, which is not always ideal. Furthermore, the model library does not give access to all the quantities.

I would also like to ask which is the reason for which these parameters are not public, is this due to some black-box model the robot is using? Is some identification performed on the robot as other users have pointed out? 

In either case, it is a pity that these parameters are not available as we now have to use another robot instead.

gollth, 2021-08-05:
Hi everyone,

starting with franka_ros version 0.8.0 there are now `<inertial>` tags inside the URDF on demand by running:

```bash
xacro $(rospack find franka_description)/robots/panda_arm.urdf.xacro gazebo:=true
```

**Please note**: These inertia values don't represent the _exact_ identified inertias of the Panda robot. Rather they come from the paper [Dynamic Identification of the Franka Emika Panda Robot With Retrieval of Feasible Parameters Using Penalty-Based Optimization](https://hal.inria.fr/hal-02265293/file/IROS_19_Panda.pdf).

However for quick prototyping these values should be good enough to simulate in Gazebo and roughly test your custom controllers. Please also check our [documentation](https://frankaemika.github.io/docs/franka_ros.html) for more details.

Since the URDF now contains inertias, I would close this issue now.

closed 2021-08-05 by gollth