frankarobotics/franka_ros #14: move_to_start Robot semantic description not found

boinappi, 2018-04-06:
Hi!
We are having a hard time to run the move_to_start.launch example.
We updated our system to 1.0.2 / 3.0.0.
`roslaunch franka_example_controllers move_to_start.launch robot_ip:=172.16.0.2 load_gripper:=false`
quits with an
> Unused args [load_gripper] for include of [..../panda_moveit.launch]

So we simply removed line 10 from the move_to_start.launch file, defining the argument load gripper for panda_moveit.launch.

After this the command runs, till it says

> You can start planning now

in green.
But nothing moves - i expect the robot to move to a predifined startpose, correct?

The trace says at
> process [move-to-start]: started with pid...
> ERROR [huge number] Robot semantic description not found. Did you forget to define or remap '/robot_description_semantic'?

As i'm not the fittest ros-guy I'm pretty stuck here. I checked filepaths twice and i guess everything is in the right place. Any ideas?

Thanks in advance!

fwalch, 2018-04-09:
Can you try the newest version of franka_ros, which is 0.4.0 (requires system 1.2.0)?

franka_ros 0.4.0 is available in the repositories, so `sudo apt install ros-kinetic-franka-ros` will set up everything for you. If you cloned the franka_ros sources from Github, you should now remove them from your catkin workspace.

boinappi, 2018-04-13:
I updated via apt, with no success.

__I solved the issue__ by building libfranka and franka_ros myself, following [this](https://frankaemika.github.io/docs/installation.html#building-from-source) tutorial.

closed 2018-04-13 by boinappi