frankarobotics/franka_ros #17: Using MoveIt's MoveGroupCommander.pick results in 'No Sampler was Constructed'

christgau, 2018-05-31:
Hi folks,

I am using MoveIt's Python interface to control a Panda. Using `set_pose_target`, `plan` and `go` works fine for moving the arm into a certain pose. However, the `pick` method does not work at all. A scene was constructed and objects were added to the scene. I tried both the `pick` method without additional grasps and using a list of grasps as well.

In the result, the error message "No sampler was constructed" is shown. Its origin is deep inside MoveIt ([source code](http://docs.ros.org/kinetic/api/moveit_ros_manipulation/html/reachable__valid__pose__filter_8cpp_source.html), line 147). It is unclear to me, what's the root cause for this problem. I am primarily developing using `panda_moveit_config/demo_launch` but I also tried it with a real robot. In both cases, the error message is the same. Is there a missing parameter in the configuration/launch files? Do I have to supply any parameter or is franka_ros/panda missing support for MoveIt's interface? Or is this issue related to a point from #13 

> The gripper didn't really work on the real robot with MoveIt, as the fingers weren't included in the panda_arm_hand group

Do I have to use the GraspAction from the franka_gripper package instead of MoveIt's methods? 

BTW: I am using version 0.4.0

PS: This is more a less a cross post of a question from [ROS answers](https://answers.ros.org/question/292280/calling-pick-results-in-no-sampler-was-constructed/)

fwalch, 2018-06-12:
I'm not sure if the problem lies with `franka_ros` or with MoveIt.. can you post this to the MoveIt issue tracker as well?

christgau, 2018-06-12:
It appeared to me that this is a Franka-specific issue. Nevertheless, I can post an issue in the MoveIt tracker but I'm not sure if I will get more replies than in ROS answers which is 0 so far... Let's have a try.

christgau, 2018-06-12:
created a new issue in the MoveIt repo: https://github.com/ros-planning/moveit/issues/948

christgau, 2018-06-18:
Although, "Panda robot is the flagship MoveIt! integration robot used in the MoveIt! tutorials" (ros-planning/panda_moveit_config@15df06ced3b0fd313301bca9f3c982c419fe452e), the version that comes with the ROS/MoveIt packages is not up to date. Using the `panda_moveit_config` from master solves the problem.

closed 2018-06-18 by christgau