frankarobotics/franka_ros #18: Simulating the Gripper in ROS

christgau, 2018-06-06:
Using the gripper in simulation appears to not be possible in a ROS environment, at the moment. Using MoveIt's pick command in simulation with the demo launch file from the panda_moveit_config package results in the 'No sampler was constructed' issue #17. 

As a replacement, I tried to use the grasp action from the `franka_gripper_node` and interfaced it from Python without problems. However, an action server must be available, obviously, but demo.launch does not provide such one. Only, the gripper node provides it but also requires a connection to the FCI and thus a real robot, to which a have only very limited access in my usecase. So, I set up a dumb (netcat) TCP server on the required port. Then the gripper node is able to start but this it fails to get the (real) gripper state which is not published via UDP as it is required for the gripper node. 

As a result, I am stuck. The only thing I can do is to move the gripper via the fake joint state publisher node from demo.launch. Anyways, there should be some way to use gripper in simulation. Just moving the robot around, which works fine with MoveIt by the way, becomes boring at some point ;-)

fwalch, 2018-06-11:
The "simulation" is provided by MoveIt's `MoveItFakeControllerManager`. The hand also has a fake controller manager, so it works the same was as the robot.

Simply running `roslaunch panda_moveit_config demo.launch` and then using the RViz MotionPlanning tab to control the `hand` (or `panda_arm_hand`) planning group with a random goal state seems to work fine, so in principle it should work.

fwalch, 2018-06-12:
Yes, you cannot use the `franka_gripper_node` (or, I think, any of the other `franka_ros` nodes) without a real robot. Even for the robot arm itself, it is only MoveIt which provides fake execution, but e.g. the `franka_control_node` topics and services are also not simulated.

christgau, 2018-06-12:
Ok. It's nice to have a confirmation for my assumption.

Are there any plans to provide kind of a simulation mode for the Franka nodes where not FCI/robot is required? In my particular use-case, I have only limited access to the robot, thus a primarily rely on simulation during development. Maybe this is true for other developers as well.

fwalch, 2018-07-19:
Sorry, I just realized I never answered you again. There are currently no plans to provide a simulation mode for the gripper or the robot. You should, however, be able to use MoveIt's fake execution with the gripper as well. Of course, then you can only use MoveIt's `control_msgs::GripperCommandAction` and not `franka_ros`s custom gripper actions.

closed 2018-07-19 by fwalch