frankarobotics/franka_ros #22: Errors by using the

krigi100, 2018-07-18:
Since I'm using Version 1.3.2 with libfranka 0.4.0 and franka_ros 0.5.0 sometimes an oscillation occurs by using the [move group interface tutorial ](https://github.com/ros-planning/moveit_tutorials/blob/kinetic-devel/doc/move_group_interface/src/move_group_interface_tutorial.cpp) with a real robot. Then the robot gets in an error like cartesian reflex or tauJ violation. 
With the older version I didn't had the problem. 
The error also occurs sometimes by planning and execute with MoveIt and the rviz gui.
Are the reason for the problem the changed rate limiters?
How can I fix the problem?.

krigi100, 2018-07-23:
Thanks a lot, 
with the velocity controller it works fine.

fwalch, 2018-08-08:
We just released a new version of libfranka and franka_ros, which should fix these problems. If you're still encountering problems, let us know and we'll reopen the issue!

closed 2018-08-08 by fwalch