frankarobotics/franka_ros #25: functionality of service calls dependes on user stop

Krebsileinchen, 2018-08-09:
I am using the latest release of franka_ros (0.6.0) with libfranka (0.5.0). When calling services from franka_control such as for example franka_control/set_load I am encountering a strange behaviour. 
The service call fails with 

> error: "libfranka: Set Load command rejected: command not possible in the current mode!"

However after closing and reopening the user stop and no further changes it works just fine.

sgabl, 2018-08-17:
I assume that the robot is in a Reflex and requires an error recovery. This can be done by closing and opening the external activation device. You can check the RobotMode in the RobotState.

Krebsileinchen, 2018-08-22:
In this case it should have been possible to call the franka control action Error Recovery instead of using the user stop, shouldn't it? However, this did not work.

sgabl, 2019-07-22:
After opening the brakes, you need to toggle the external activation device once. This won't be necessary in an upcoming release! If this happens after a reflex, try to call the franka control action Error Recovery, download directly after the logs of the robot and send them to support@franka.de.

closed 2019-07-22 by sgabl