frankarobotics/franka_ros #27: Missing Controller for the Gripper

christgau, 2018-08-16:
In relation to #26 and ros-planning/panda_moveit_config#9 I'd like to add that I was able to use the gripper with MoveIt's pick and place API only when I added a definition for the controller and its joints including timing and goal constraints. The controller must have the same name as in `panda_gripper_controllers` (taking https://github.com/ros-planning/panda_moveit_config/pull/9/commits/a9fa9dce22ce5765af030d3c712683f149f1eb5e#diff-3ea2eda6749be8544a6f913da5bacd23 into account). I used a position_controllers/JointTrajectoryController. 

For getting things work, I added the according changes to `default_controllers.yaml` of the `franka_control`package. I assume that this is not the perfect place, but it worked for me.

I'd appreciate if the according changes in the correct file would be integrated in the franka_ros package  such that the gripper is usable with MoveIt and a real robot.

fwalch, 2018-08-17:
If you run `roslaunch franka_gripper franka_gripper.launch`, it will run a `franka_gripper` node with a `gripper_action`. This is exactly what's required by https://github.com/ros-planning/panda_moveit_config/commit/a9fa9dce22ce5765af030d3c712683f149f1eb5e#diff-3ea2eda6749be8544a6f913da5bacd23, so this should work already. I only tested with RViz, not with the API, however.

In the newest version, `franka_control` will run the gripper node automatically if `load_gripper:=true` is given to `roslaunch`.

christgau, 2018-08-17:
The introduction of auto-starting is a nice feature.

I might be mistaken, but I don't think that the Franka-specific `gripper_action` is used by MoveIt for controlling the gripper. From my understanding, the joints and the responsible controller type have to be specified in order to use them with MoveIts `pick` routine and the associated [Grasp messages](http://docs.ros.org/api/moveit_msgs/html/msg/Grasp.html). If that is correct then I suggest to add the required configuration to franka_ros in order to enable the usage of the gripper.

fwalch, 2018-08-17:
`gripper_action` is implemented with the standard `control_msgs/GripperCommandAction`, not one of the Franka-specific types. This is also how it's set up in the [tutorial](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/controller_configuration/controller_configuration_tutorial.html).

I'm not sure how all of that relates to the `pick` routine, though.. I'll look into it.