frankarobotics/franka_ros #30: Grasping fails even though it succeeds

aPonza, 2018-10-01:
I implemented a simple action client with:
<pre>
actionlib::SimpleActionClient<franka_gripper::GraspAction> ac_gg("/franka_gripper/grasp", true);
franka_gripper::GraspGoal goal_gg;
franka_gripper::GraspResult result_gg;
goal_gg.force = 10; // N
goal_gg.speed = 0.05; // m/s
goal_gg.width = 0.036; // m
goal_gg.epsilon.inner = 0.005; // m?
goal_gg.epsilon.outer = 0.005; // m?

(...)

bool state = result_gg.success;
ROS_INFO("The action finished with status: %u", state);
</pre>

and when I run it, attempting to grasp an object that satisfies the specs sent as a goal, `rosrun panda_moves grasping_test_client_node` returns:
<pre>
[ INFO] [1538150133.985779942]: Waiting for the action server to start.
[ INFO] [1538150134.273582728]: The action server is live, sending goal.
[ INFO] [1538150136.121196455]: <b>The action finished with status: 0</b>
</pre>

while `rostopic echo /franka_gripper/grasp/result` returns:
<pre>
header: 
  seq: 14
  stamp: 
    secs: 1538150136
    nsecs: 120925357
  frame_id: ''
status: 
  goal_id: 
    stamp: 
      secs: 1538150134
      nsecs: 273606745
    id: "/test_grasping-1-1538150134.273606745"
  status: 3
  text: ''
result: 
  <b>success: True</b>
  error: ''
---
</pre>

As a matter of fact the object is held in place by the gripper, and sending a stop action stops the 10N to be exerted and I can remove the object easily from the grasp.

Am I missing something?

As a side note, epsilon seems to represent the grasp width error, but is it to be considered as meters (5mm), or is it a percentage error (0.5%)?

fwalch, 2018-10-01:
As the success flag seems to be correctly returned from `franka_ros`, I don't think there should be any problem there. Also the gripper behavior you describe sounds correct. Maybe there is some bug in your omitted code "(...)"?

> As a side note, epsilon seems to represent the grasp width error, but is it to be considered as meters (5mm), or is it a percentage error (0.5%)?

It's in meters (see http://docs.ros.org/kinetic/api/franka_gripper/html/msg/GraspEpsilon.html)

aPonza, 2018-10-01:
This is the complete node:

<pre>
#include "ros/ros.h"

#include "actionlib/client/simple_action_client.h"
#include "franka_gripper/GraspAction.h"

int main(int argc, char **argv)
{
  ros::init(argc, argv, "test_grasping");

  // create the action client, the goal, and wait for the server to be live
  actionlib::SimpleActionClient<franka_gripper::GraspAction> ac_gg("/franka_gripper/grasp", true);
  franka_gripper::GraspGoal goal_gg;
  franka_gripper::GraspResult result_gg;
  goal_gg.force = 10; // N
  goal_gg.speed = 0.05; // m/s
  goal_gg.width = 0.036; // m
  goal_gg.epsilon.inner = 0.005; // m
  goal_gg.epsilon.outer = 0.005; // m

  ROS_INFO("Waiting for the action server to start.");
  ac_gg.waitForServer();
  ROS_INFO("The action server is live, sending goal.");

  // grasp the object
  ac_gg.sendGoal(goal_gg);

  // wait for the action to return then print what happened
  bool finished_before_timeout = ac_gg.waitForResult(ros::Duration(5.0));

  if (finished_before_timeout)
  {
    actionlib::SimpleClientGoalState state_ac = ac_gg.getState();
    ROS_INFO("The action client reports the status: %s",state_ac.toString().c_str());

    bool state_gg = result_gg.success;
    ROS_INFO("The grasp %s.", state_gg ? "SUCCEEDED" : "FAILED");
  }
  else
    ROS_INFO("The action did not finish before the time out.");

  return 0;
}
</pre>

I changed a bit the output but the problem remains. I tried to declare the result right before the printout in case it was an "update" issue but it doesn't seem to change a thing.

aPonza, 2018-10-04:
The correct way isn't result_gg.success but ac_gg_.getResult()->success, as result_gg was never actually assigned the proper GraspResult value (even if declaring it before the printout, because that's assignment vs. declaration).

closed 2018-10-04 by aPonza