frankarobotics/franka_ros #32: current_errors doesn't keep the error state once the arm stops

aPonza, 2018-10-17:
I can't find a way to tell if the arm is in an error state:
- since the robot stops the execution once a `franka::Errors error` happens, `rostopic echo /franka_state_controller/franka_states/current_errors` reports all flags as false (e.g. cartesian_velocity_violation will of course not be True now that the robot is stopped, same for joint_reflex or cartesian_reflex for example, but this is not a problem for self_collision as that stops the robot in an error position);
- `rostopic echo /franka_state_controller/last_motion_errors` reports the last error as a True, since that's what stopped the robot, however, it doesn't update after the error_recovery action is called and,  logically, it shouldn't;

This combination blocks the possibility to automatically recover from an error and execute a different operation completely. I was expecting current_errors to keep the error that's currently blocking the execution of any movement until an error recovery is performed.

Is this intended behaviour and/or can this be changed? Is there a different way to get the current error state of the robot that I'm not seeing?

sabaghian, 2018-10-17:
Maybe robot_mode is what u need [Robot_mode](https://frankaemika.github.io/libfranka/structfranka_1_1RobotState.html#a4943ae75e0e2ec534e0afac31cbcc987).

fwalch, 2018-10-17:
You can get the robot's mode from libfranka's [`RobotState::robot_mode`](https://frankaemika.github.io/libfranka/structfranka_1_1RobotState.html#a4943ae75e0e2ec534e0afac31cbcc987), but it looks like we forgot to expose that over a ROS topic. We'll add that with a future release. In the meantime, you can use a custom ros_control controller which takes a `franka_hw::FrankaStateInterface` (like the [`FrankaStateController`](https://github.com/frankaemika/franka_ros/blob/kinetic-devel/franka_control/include/franka_control/franka_state_controller.h)) and directly read the mode from the libfranka `RobotState` (and maybe publish from there).

aPonza, 2018-10-19:
All right, super, cheers!

fwalch, 2018-10-22:
@aPonza I pushed an update to the `develop` branch where the `FrankaStateController` publishes the robot mode.

aPonza, 2018-10-22:
Thanks, https://github.com/frankaemika/franka_ros/commit/3991f55ffc9c1f4928bbbad03b452f4202e96eea solved my issue.

The only thing I'm not sure is intended is if I `roslaunch panda_moveit_config panda_moveit.launch` the mode changes from ROBOT_MODE_IDLE to ROBOT_MODE_MOVE, even if panda itself is idle. Else this can be closed.

fwalch, 2018-10-22:
Yes, this is intended behavior. When you launch MoveIt, the joint trajectory controller will become active and control the robot (to hold its current position), so from the robot's point of view, it isn't idle.

closed 2018-10-22 by fwalch

aPonza, 2019-01-09:
I noticed today that the last two comments apply when `roslaunch`ing while in reflex mode, which results in message `[ERROR] - [/franka_control::main::206]: libfranka: Move command rejected: command not possible in the current mode!`.

There appears to be no command sent to the robot at all, so it must be the takeover from the joint trajectory controller. Leaving this here for anybody else that might stumble upon it.