frankarobotics/franka_ros #33: Stopping example controllers

bomarali, 2018-10-17:
Hello everyone!

I have problem with stopping joint_position_example_controller. I have tried running the "kill" command from the controller_manager kill both the joint_position_example_controller as well as the franka_state_controller (or something like that I don't remember the exact name). But after I try to re-launch the same joint_position_example_controller I get the "Move command rejected: command not possible in the current mode!" error. Could you please tell me what am i doing wrong? (Frankly I am new to ROS)

fwalch, 2018-10-18:
If the controller is just killed and not terminated properly, the robot will end up in reflex mode. You can run [error recovery](https://frankaemika.github.io/docs/franka_ros.html#franka-control) to get the robot back into idle mode. Note that the robot mode is currently not published into ROS, but we'll fix that (see https://github.com/frankaemika/franka_ros/issues/32#issuecomment-430643111).

fwalch, 2018-10-22:
Yes, I think that should work. See also the [documentation](http://docs.ros.org/melodic/api/controller_interface/html/c++/classcontroller__interface_1_1ControllerBase.html#aa694bd08e937c21e3011725f469adf94).

bomarali, 2018-10-23:
I have seen that the joint_velocity_example_controller.cpp does have the stopping method with following comment:
  // WARNING: DO NOT SEND ZERO VELOCITIES HERE AS IN CASE OF ABORTING DURING MOTION
  // A JUMP TO ZERO WILL BE COMMANDED PUTTING HIGH LOADS ON THE ROBOT. LET THE DEFAULT
  // BUILT-IN STOPPING BEHAVIOR SLOW DOWN THE ROBOT.

what is the built-in stopping behavior?

bomarali, 2018-11-09:
I have done some further testing. To sum up, I still don't know what exactly causes controllers to start/stop (no updates in between) on spawn before going to the standard start/update until not stopped. I have rewritten the joint_velocity_example_controller to receive target velocities which in turns allows me to gently reduce joint_velocity to zero or rather nearly non-zero values (the bag of velocities show some noise, I guess it is to be expected). At this point it was safe to stop the controller through rosservice call /controller_manager/switch_controllers service.

closed 2018-11-09 by bomarali

fwalch, 2018-11-09:
>  still don't know what exactly causes controllers to start/stop (no updates in between) on spawn

Sorry, I forgot to respond here. I can clarify this at least: when starting a motion or when re-starting after recovering from an error, the controllers are currently [reset](https://github.com/frankaemika/franka_ros/blob/3f695940b95af8b72fcefc2f65ec6f31da5a1b80/franka_control/src/franka_control_node.cpp#L139-L141), which is what's causing this.

bomarali, 2018-11-09:
This is a bit strange to me. I got a similar explanation from ros_control group, at which point I went through the log of execution times which showed that the reset of the franka_control_node happens after start stop of my custom controller. This led me to a conclusion that reset is not related to the start stop behavior. But if you say that it is, could time stamps in the log be wrong?

fwalch, 2018-11-12:
I don't think that it's very likely that the time stamps are wrong, but maybe the logging for the reset happens after the controllers are restarted instead of before? Which log entry did you use to determine the "reset time"?

bomarali, 2018-11-12:
I am pretty sure that I have first just added 2 ROS_INFOs around the reset command to looked how their timing relates to ROS_INFOs I had in starting and stopping. I am not sure if it is scientific enough though. I also remember that the # of the message in the rqt_console did not correspond to time stamps on them, i.e. some messages were assigned # not in chronological order. I have then went into the actual log files but I do not remember if I got anything substantial from them.