frankarobotics/franka_ros #34: Release to ROS Melodic LTS

davetcoleman, 2018-10-18:
MoveIt! is making a big push for Melodic right now and our tutorials do not work as well without franka_ros released:
https://github.com/ros-planning/moveit/pull/1098

davetcoleman, 2018-10-18:
Thanks for the quick reply!

Could we get the franka_description released any earlier? Because its the example robot for all of MoveIt!, our Melodic tutorials are currently broken and our CI doesn't pass.

gavanderhoorn, 2018-10-18:
Related ROS Answers post: [Moveit! tutorials on ROS Melodic](http://answers.ros.org/question/303278).

fwalch, 2018-10-19:
> Could we get the franka_description released any earlier?

Sure! See release PR here: https://github.com/ros/rosdistro/pull/19307

kargarisaac, 2019-05-11:
Hi,
Isn't there any melodic version for franka_ros and libfrabka yet?

ZakariaChekakta, 2019-07-18:
Is there in new about Melodic verison?

fwalch, 2019-07-18:
Building `libfranka` and `franka_ros` from source with Ubuntu 18.04/ROS Melodic has been possible since libfranka 0.6.0, where we fixed some issues related to newer versions of GCC and Doxygen.
As for packaging for ROS, this has been merged recently (https://github.com/ros/rosdistro/pull/21739, https://github.com/ros/rosdistro/pull/21715) and should show up in the repositories soon (see [here](http://repositories.ros.org/status_page/melodic_default.html?q=franka) for status). Sorry for taking so long.

closed 2019-07-18 by fwalch