frankarobotics/franka_ros #35: Real-time joint space control

bomarali, 2018-10-31:
Hi all!
I need some information on how the joint space motion commands get executed. For instance in the joint_position_example_controller the joint motion is achieved by set_command on hardware_interface::JointHandle. I am interested to know how the target is achieved, i.e. what joint speeds and accelerations get used. The only hint I have found was in the defaul_controllers.yaml goal time constraint of 0.5. Does that mean that desired joint position must be achieved within 0.5 sec?

fwalch, 2018-10-31:
Each `libfranka` controller needs to return values with 1KHz. This is also explained in more detail [here](https://frankaemika.github.io/docs/libfranka.html#realtime-commands).

We provide [ros_control](http://wiki.ros.org/ros_control) integration with `libfranka` with `franka_control`. The `joint_position_example_controller` is implemented with ros_control and also runs with 1KHz. The speed and acceleration is defined by the values the controller commands for the next ms.

The 0.5 in `default_controllers.yaml` is a parameter of the `joint_trajectory_controller`, which is a generic ros_control controller and not part of `franka_ros`. See the documentation [here](http://wiki.ros.org/joint_trajectory_controller#Parameters).

bomarali, 2018-10-31:
Thank you for the clarification on the default_controllers.yaml part.
I am still unfortunately in the dark on the matter speeds and accelerations. Could you please tell me how can I specify joint velocities and accelerations or target time for joint position command with ROS. I know that there are commands for sending joint velocities but I assume that they would simply override joint position commands, won't they?

fwalch, 2018-11-06:
If you implement a controller, the values you set as the robot's next joint position goal for the next millisecond determine the velocity and acceleration.

The `joint_trajectory_controller`, for example, receives a trajectory with desired velocities (http://docs.ros.org/melodic/api/trajectory_msgs/html/msg/JointTrajectoryPoint.html) and then outputs the next joint position accordingly.

bomarali, 2018-11-09:
I have some problems with "communication_constraints_violation" when idle. I have made some adjustments to the joint_velocity_example_controller so that it subscribes to target delta_theta joint motions that I publish from outside and moves accordingly. Whenever the target is reached I send zero velocities. However it results in communication_constraints_violation. I have noticed that in joint_position_example_controller if I dont send any position updates the robot would go to reflex, however it would be fine if I keep sending the same position (I was just running the controller on idle to check publications/subscriptions). Yet it doesn't seem to be the case with joint_velocity control. I have done ping check, results are similar to one in documentation. I have also performed the advanced communication check which gave 59 lost robot_states in 10000 milliseconds. I have disabled cpu frequency scaling, full battery, plugged in. I have also deleted all ROS_INFO calls from the controller so that printf doesnt eat execution time. Overall I am surprised that robot works well when given perpetual motion but runs into a communication error when idle.

fwalch, 2018-11-12:
> I am surprised that robot works well when given perpetual motion but runs into a communication error when idle.

I am surprised to hear this, too. I adjusted the `joint_velocity_example_controller` to set zero velocities, but I don't see any issues when running it.

bomarali, 2018-11-12:
You are actually right. I have also set velocities to zero and get no error. I do however get zeros if I do this:

void JointVelocityExampleController::update(const ros::Time& /* time */,
                                            const ros::Duration& period) {
  if (elapsed_time_.toSec() < time_max)  {
    for (int j = 0; j < 7; j++)    {
      omega[j] = omega_max[j] * std::sin(M_PI / time_max * elapsed_time_.toSec());
    }
  }
  else   {
    for (int j = 0; j < 7; j++)     {
      omega[j] = 0.0;
    }
  }
  elapsed_time_ += period;
  for (int j = 0; j < 7; j++)   {
    velocity_joint_handles_[j].setCommand(omega[j]);
  }
}

In short I sent this message rostopic pub --once joint_velocity_example_controller/dummy_joint_velocity franka_example_controllers/buki "{delta_theta:[0.0,0.0,0.0,0.0,0.2,0.3,0.3]}" which on callback calculates joint_velocity trajectory as a half sine that would move the robot for the desired delta angles within maximum allowed time which resets during callback. However shortly after the completion of the command the robot would go into the reflex mode due to communication error 

void JointVelocityExampleController::callback(const franka_example_controllers::buki::ConstPtr& msg) {
  if (elapsed_time_.toSec() > time_max)   {
    for (int i = 0; i < 7; i++)  {
      delta_theta_container[i] = msg->delta_theta[i];
      elapsed_time_ = ros::Duration(0.0);
      omega_max[i] = delta_theta_container[i] * M_PI / (2 * time_max);
    }
  }
}

Gracedgl, 2019-02-22:
I was wondering where I could get the Jaccobi matrix(J) and inertia matrix (M(theta)) ,I couldn't find it in franka ros cpp files.Thank you

markusgft, 2019-03-11:
@fwalch @bomarali I would like to follow up on this issue - I actually observe the behaviour described by @bomarali on our Panda robot. When I send zero velocities at the end of an (arbitrarily) long trajectory, I obtain a communication constraints violation.
 * I can replicate it with official ROS control trajectory controllers
 * It is highly undeterministic (i.e. it sometimes is gone after a system restart, sometimes it only starts to show up after hours of operation). 
 * I observe it not only at the velocity interface, but also at the position interface (some issue at trajectory end).
Is there any update on this topic?

jacknlliu, 2019-04-28:
The current franka ros hardware interface node for ros controller can't work correctly.

berktepebag, 2019-07-05:
I can confirm that sending 0 velocities with velocity control causes communication error. Instead I sent 0.000001 velocity and it's working fine. I did not have such problem with position controller.

markusgft, 2019-12-05:
Dear @fwalch, I would like to push this to the front of your queue -- the above described failure 
```
When I send zero velocities at the end of an (arbitrarily) long trajectory, I obtain a communication constraints violation.
```
still occurs randomly. We have observed it on different franka robots in different locations with different firmware versions. 
I believe the franka customers in this thread would greatly appreciate a response, a clarification of the issue and a note on when this is going to be fixed. Thanks in advance.

carlosjoserg, 2020-03-06:
> I can confirm that sending 0 velocities with velocity control causes communication error. Instead I sent 0.000001 velocity and it's working fine. I did not have such problem with position controller.

Confirming this comment as well. Sending veclocity 0.0 rises the error, sending velocity 0.000001 is ok, and normal operation continues.

fe-support, 2023-06-20:
Closing due to inactivity. Please open this git issue again if the topic still exists.

closed 2023-06-20 by fe-support