frankarobotics/franka_ros #41: Using CXX_STANDARD requires CMake 3.1

aPonza, 2018-12-14:
Compilation of my own packages depending on franka packages has always worked. At the same time Eclipse has always highlighted `#include <franka_(...)` directives as unresolved dependencies. I decided to dig into it and solved it by changing usage of `add_compile_options(-std=c++14)` like in the [boilerplate CMakeLists.txt](https://github.com/ros-infrastructure/catkin_pkg/blob/master/src/catkin_pkg/templates/CMakeLists.txt.in) that's provided with `catkin_create_pkg` with your `set(CMAKE_CXX_STANDARD 14) set(CMAKE_CXX_STANDARD_REQUIRED ON)`.

This is when I found out the `CXX_STANDARD` property was [introduced in CMake 3.1](https://cmake.org/cmake/help/v3.1/release/3.1.0.html#properties) so you should most likely have `cmake_minimum_required(VERSION 3.1)` in your CMakeLists.txt (or set a property like in this related [Q&A](https://stackoverflow.com/questions/37621342/cmake-will-not-compile-to-c-11-standard#comment62725635_37621342)).

fwalch, 2018-12-14:
Thanks, looks like an oversight in `franka_ros`. `libfranka` already requires CMake 3.4.

fwalch, 2019-01-08:
Pushed a fix to `develop`.

commit 3975f8e5 (2019-01-07): Update minimum required CMake version

Resolves #41
closed 2019-01-08 by fwalch