frankarobotics/franka_ros #44: Can I simulate franka panda in Gazebo

xdk-rachel, 2019-01-03:
Can I simulate franka in Gazebo?

erdalpekel, 2019-01-14:
Hi
I have written a [blog post](https://erdalpekel.de/?p=55) about the integration of panda robot into gazebo.
Would appreciate a star on my [repo](https://github.com/erdalpekel/panda_simulation)
thanks

mehditlili, 2019-01-15:
@erdalpekel you could maybe consider doing a pull request to the main repo, if your changes  are compatible.

erdalpekel, 2019-01-15:
@mehditlili I will consider this if it can be done without affecting the repos. I will check that.

JensPeter92, 2019-03-08:
@erdalpekel I have tested your gazebo model with MoveIt! linkage and it is working perfectly fine. You have implemented the standard ros controller, but is it also possible to try out the Franka Panda example controllers (e.g. cartesian impedance controller) ?.

I mean the Hardware framework is different to the standard ros::effortcontroller and SimpleTransmission, so do i have to completely change the transmission communication and also add additional sensors like torque sensors in the gazebo model, in order to obtain the wrenches?

erdalpekel, 2019-03-18:
@JensPeter92 sorry for the late response.
The question is not _if_ the controller will work. It is more the fact that the gazebo_ros plugin and gazebo's hardware interface that polls the robot state from the dynamics engine will likely not be compatible.

I've had a look at one of the example controllers `elbow_example_controller.cpp` where in line 40 it tries to receive the robot state using a proprietary interface:

```
auto state_interface = robot_hardware->get<franka_hw::FrankaStateInterface>();
```

If you were to rewrite this to use one of the interfaces that the standard ROS controllers use, e.g. `EfortJointInterface` or `VelocityJointInterface`, it would be possible to use the Franka controllers.

Here is a drawing from gazebo control tutorials:
[image]

The problem is in the connection between the top-left box and yellow box in the middle.

I would strongly suggest that you clone the repositories 

- [ros_controllers](https://github.com/ros-controls/ros_controllers.git)
- [gazebo_ros_pkgs](https://github.com/ros-simulation/gazebo_ros_pkgs.git) - `gazebo_ros_control`
- [gazebo](https://bitbucket.org/osrf/gazebo) - namespace `physics`

and get familiar with what is happening under the hood.

padmaja-kulkarni, 2020-11-30:
Hi! I was wondering if there is any update on using, for example, Cartesian impedance controller on Gazebo, with Franka Panda?

Thanks in advance.

gollth, 2021-08-05:
Hello and sorry for the late reply. Thanks @erdalpekel, your blog was an inspiration for the following:

We are happy to announce that with version 0.8.0 of franka_ros there is Gazebo integration for the panda. Next to the necessary URDF changes (like adding estimated inertas) this includes also:

* Gravity Compensation in Gazebo, similar to how the real Panda does it
* Offering the same `FrankaState`- & `FrankaModelInterface` as `franka_control` does
* Offering the same action interfaces like `franka_gripper` does
* Offering similar service (non-realtime) interfaces similar to how `franka_control` does it

Hopefully this will allow you to simulate existing ROS controllers now in Gazebo as well. Now you can use the the Cartesian Impedance Example Controller without any code changes also in Gazebo.

Check out our [documentation](https://frankaemika.github.io/docs/franka_ros.html#franka-gazebo) for more information.

Since this feature is now implemented I would close this issue now.

closed 2021-08-05 by gollth