frankarobotics/franka_ros #45: Panda C++ IK Solver?

davetcoleman, 2019-01-15:
Does Franka Emika provide, or is anyone aware of, an IK solver we can use locally in our motion planning code, as opposed to hidden in Franka's control software? An customized/optimized IK solver for Panda would greatly speed up MoveIt! integrations and other control needs of our clients. Of course I already know about KDL and IKFast, but IKFast has issues working with Panda according to @mlautman.

Thanks!

fwalch, 2019-01-17:
Unfortunately, we don't provide an IK solver.

gavanderhoorn, 2019-01-17:
@fwalch wrote:
> Unfortunately, we don't provide an IK solver.

Is there a specific reason for that?

jrmout, 2019-01-17:
@mlautman Could you elaborate on the issues of IKFast? Thanks!

closed 2019-07-05 by sgabl

gavanderhoorn, 2019-07-05:
No comment from Franka Emika on this?