frankarobotics/franka_ros #46: Virtual Wall requires more details

Adito5393, 2019-01-16:
Could you provide more details on how to define the virtual wall? How does it work?

I tried setting the p_frame as an identity matrix (1 on diagonal, rest 0) and "object_world_size" set to x=0.6 y=0.3 z=0.5. I did not encounter any virtual wall while moving around the "object_world_size" position using the cartesian impedance controller example.
[Link API](https://frankaemika.github.io/libfranka/structfranka_1_1VirtualWallCuboid.html)

djl11, 2019-07-04:
Is there any plan to add this virtual wall feature? There is also research_interface::robot::GetCartesianLimit defined in libfranka API, is this limit not functional in any way at the moment?

sgabl, 2019-07-05:
At the moment we can not provide a time-frame for the virtual wall FCI implementation.

closed 2019-07-05 by sgabl