frankarobotics/franka_ros #47: Compilation with gcc-8 leads to non-working franka_ros

aPonza, 2019-01-18:
As part of enabling some C++17 functions in my own code I needed to upgrade gcc as the shipped version with 16.04 is gcc-5 which doesn't support it. Looking at [cppreference](https://en.cppreference.com/w/cpp/compiler_support#C.2B.2B17_features) I can see if I want to have everything from C++17 I need gcc-8. Looking at [Porting to GCC 8](https://www.gnu.org/software/gcc/gcc-8/porting_to.html) to see if there might be issues I figured there were only "nice new things" for me, so nothing that should break anything on my side.

I cleaned the workspace and recompiled after the installation and update-alternatives switch to gcc-8, saw a passing warning on one of the franka_hw functions (from -Wreturn-type, one of the new things in gcc-8, makes sense) but nothing more. The warning is:

```
Warnings   << franka_hw:make /home/ap/catkin_ws/logs/franka_hw/build.make.000.log                   
/home/ap/catkin_ws/src/franka_ros/franka_hw/src/franka_hw.cpp: In function ‘std::ostream& franka_hw::{anonymous}::operator<<(std::ostream&, franka::ControllerMode)’:
/home/ap/catkin_ws/src/franka_ros/franka_hw/src/franka_hw.cpp:25:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
```

However, upon successful compilation, `roslaunch`ing my process fails.

Installing gcc-7 and recompiling shows no warnings and yields a working application, and since the feature I wanted for myself is included already in gcc-7, I'm fine with that at the moment.

I'm assuming it's franka_ros that's failing because my package only depends on source installs from franka_ros and panda_moveit_config, but the latter has no source to compile; all other dependencies are installed from binary, libfranka is not being touched even though it would be a source install as well.

Launching separately the nodes that panda_control_moveit_rviz.launch launches I see this succession of events:
- franka_control.launch seemingly launches correctly;
- panda_moveit.launch launches with an exception thrown

```
[... omitting the summary for brevity ...]

process[controller_spawner-1]: started with pid [10885]
process[move_group-2]: started with pid [10886]
[ INFO] - [/move_group::core::RobotModel::buildModel::93]: Loading robot model 'panda'...
[ INFO] - [/move_group::core::RobotModel::buildModel::93]: Loading robot model 'panda'...
[ INFO] - [/move_group::PlanningSceneMonitor::startPublishingPlanningScene::331]: Publishing maintained planning scene on 'monitored_planning_scene'
[ INFO] - [/move_group::main::210]: MoveGroup debug mode is OFF
Starting context monitors...
[ INFO] - [/move_group::PlanningSceneMonitor::startSceneMonitor::964]: Starting scene monitor
[ INFO] - [/move_group::PlanningSceneMonitor::startSceneMonitor::970]: Listening to '/planning_scene'
[ INFO] - [/move_group::PlanningSceneMonitor::startWorldGeometryMonitor::1041]: Starting world geometry monitor
[ INFO] - [/move_group::PlanningSceneMonitor::startWorldGeometryMonitor::1057]: Listening to '/collision_object' using message notifier with target frame '/panda_link0 '
[ INFO] - [/move_group::PlanningSceneMonitor::startWorldGeometryMonitor::1072]: Listening to '/planning_scene_world' for planning scene world geometry
[ INFO] - [/move_group::PointCloudOctomapUpdater::start::114]: Listening to '/camera/depth_registered/points' using message filter with target frame '/panda_link0 '
[ INFO] - [/move_group::PlanningSceneMonitor::startStateMonitor::1134]: Listening to '/attached_collision_object' for attached collision objects
Context monitors started.
[ INFO] - [/move_group::OMPLInterface::OMPLInterface::55]: Initializing OMPL interface using ROS parameters
[INFO] - [/controller_spawner::main::116]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] - [/controller_spawner::main::121]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] - [/controller_spawner::main::129]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] - [/controller_spawner::main::184]: Loading controller: position_joint_trajectory_controller
[ INFO] - [/move_group::PlanningPipeline::configure::119]: Using planning interface 'OMPL'
[ INFO] - [/move_group::FixWorkspaceBounds::FixWorkspaceBounds::53]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] - [/move_group::FixStartStateBounds::FixStartStateBounds::60]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] - [/move_group::FixStartStateBounds::FixStartStateBounds::65]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] - [/move_group::FixStartStateCollision::FixStartStateCollision::57]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] - [/move_group::FixStartStateCollision::FixStartStateCollision::68]: Param 'jiggle_fraction' was set to 0.05
[ INFO] - [/move_group::FixStartStateCollision::FixStartStateCollision::73]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] - [/move_group::PlanningPipeline::configure::163]: Using planning request adapter 'Add Time Parameterization'
[ INFO] - [/move_group::PlanningPipeline::configure::163]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] - [/move_group::PlanningPipeline::configure::163]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] - [/move_group::PlanningPipeline::configure::163]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] - [/move_group::PlanningPipeline::configure::163]: Using planning request adapter 'Fix Start State Path Constraints'
[INFO] - [/controller_spawner::main::192]: Controller Spawner: Loaded controllers: position_joint_trajectory_controller
Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 207, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/kinetic/lib/controller_manager/spawner", line 199, in main
    resp = switch_controller(loaded, [], 2)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 435, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 525, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
[INFO] - [/controller_spawner::shutdown::55]: Shutting down spawner. Stopping and unloading controllers...
[INFO] - [/controller_spawner::shutdown::64]: Stopping all controllers...
[... 25 lines cut ...]
* MoveGroup using: 
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] - [/move_group::MoveGroupContext::status::84]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] - [/move_group::MoveGroupContext::status::85]: MoveGroup context initialization complete

You can start planning now!

```

- at the same time the exception is thrown, I get `REQUIRED process [franka_control-2] has died!` on the other tab;
- my process dies as soon as code hits the first move command.

Please let me know if I am mistaken in my assumptions or if I can help in some other way, I'd be happy to.

fwalch, 2019-01-18:
This is very interesting, thank your for reporting this. Does it help if you fix the warning (i.e. add `return ostream;` to the function)?

aPonza, 2019-01-18:
Seems like the answer could be a no, but it might be my fault here:
- compilation succeeds with no further warnings;
- instead of the exception, panda_moveit.launch reads correctly:

```
[INFO] - [/controller_spawner::main::192]: Controller Spawner: Loaded controllers: position_joint_trajectory_controller
[INFO] - [/controller_spawner::main::201]: Started controllers: position_joint_trajectory_controller
[ INFO] - [/move_group::MoveItSimpleControllerManager::MoveItSimpleControllerManager::146]: Added FollowJointTrajectory controller for position_joint_trajectory_controller
[ INFO] - [/move_group::MoveItSimpleControllerManager::MoveItSimpleControllerManager::137]: Added GripperCommand controller for franka_gripper
[ INFO] - [/move_group::MoveItSimpleControllerManager::getControllersList::197]: Returned 2 controllers in list
```

- starting my own process works (the robot moves to 'ready') until homing, where I get an infinite loop of:

```
[ INFO] - [/<my_node>::true_type{})) <my_ns>::internal::printResultIfSuccessExists::50]: The franka_gripper/HomingAction SUCCEEDED.
```

Commenting out the homing makes my process successfully execute the next move until it's time to grasp, where I get an infinite loop of:

```
[ INFO] - [/<my_node>::true_type{})) <my_ns>::internal::printResultIfSuccessExists::50]: The franka_gripper/GraspAction FAILED.
```

Now comes an interesting bit: uncommenting the homing and commenting out my template for the print statement (i.e. I never print/return if an action's `ResultConstPtr` doesn't have a `success` member, factually ignoring the action's result) I have a homing loop, where the grippers open and close indefinitely. The same happens with recommenting the homing bit, just for the grasping (so I see a loop for the rest of the output from the action client, but the fingers stay closed and the robot idles there).

No clue as to how I could proceed from here, let me know if I can help in any other way.

aPonza, 2019-05-15:
Not sure what was wrong at that time, but adding `return ostream;` does fix the issue.

closed 2019-05-15 by aPonza