frankarobotics/franka_ros #48: Moving gripper using Moveit and Rviz

akashcastelino, 2019-01-24:
Hi,

In demo.launch when using the 'panda_arm' move group, I can plan a motion using the interactive arrow markers.

But when I use the 'hand' move group, I am not able to interact with the hand to open/close the hand fingers.

P.s. For the 'hand' move group, in Planning>Select Goal Pose when I choose 'Random' and Execute, the grippers do move, so I think it should be possible to also manually move the gripper fingers but I'm not sure how why there is no interactive markers to move the gripper fingers.

fwalch, 2019-01-25:
The MoveIt config is now maintained at https://github.com/ros-planning/panda_moveit_config, so it's probably better to ask your question there. I am not sure if MoveIt even supports this kind of interaction with grippers from RViz. Otherwise, the MoveIt config needs to be fixed somehow.

FWIW, with a real robot, you can use several actionlib actions to control the gripper with a [franka_gripper_node](http://wiki.ros.org/franka_gripper).

closed 2019-01-25 by fwalch

gavanderhoorn, 2019-01-25:
To keep things connected: I believe this is a cross-post of [Open/Close end effector with Moveit Rviz](http://answers.ros.org/question/313637) on ROS Answers.