frankarobotics/franka_ros #55: Franka Gazebo Model with Franka Example Control

JensPeter92, 2019-03-11:
Hi,

Recently, I have implemented the Franka robot in Gazebo as described in https://erdalpekel.de/.
The robot is controllable, but the implemented controllers are analogous to the universal ROS-controller.
In order to use the Franka example controller, the Franka Hardware Interface needs to be implemented. 
Is there any guide to transfer the Panda controllers to the gazebo model?
I mean urdf model is known, but only the low-level control with transmission needs to be adjusted. Is there any previous work or guide?

Islamrial, 2019-06-25:
Hi, hope you doing well
I'm trying to do the same think, so if you  done it or you have any help please give me feedback please.
Thank you in advance

wobin0, 2019-12-17:
Hello!
I am aswell trying to do the same thing. Is their already further work done on this?
In the end i want to a apply joint torques to a gazebo model and read the joint torques. For this I need the Franka hardware interface.
Thanks in advance.

gollth, 2021-08-05:
Hello and sorry for the late reply.

We are happy to announce that with version 0.8.0 of franka_ros there is Gazebo integration for the panda. Next to the necessary URDF changes (like adding estimated inertas) this includes also:

* Gravity Compensation in Gazebo, similar to how the real Panda does it
* Offering the same `FrankaState`- & `FrankaModelInterface` as `franka_control` does
* Offering the same action interfaces like `franka_gripper` does
* Offering similar service (non-realtime) interfaces similar to how `franka_control` does it

Hopefully this will allow you to simulate existing ROS controllers now in Gazebo as well.

Check out our [documentation](https://frankaemika.github.io/docs/franka_ros.html#franka-gazebo) for more information.

closed 2021-08-05 by gollth

Yitian-Xian, 2023-05-08:
@gollth Hi, may I know if gravity compensation is always being implemented automatically? For example, when I am learning the cartesian_impedance_example_controller, what I can see is tau_d << tau_task + tau_nullspace + coriolis, which are from Cartesian PD control, nullspace PD control, and the coriolis effect. I am very confused about this question. Could you please help me with it? Thank you.