frankarobotics/franka_ros #56: Need some clarity on various force data in franka state

bomarali, 2019-03-22:
I have modified the cartesian impendence example controller to run in a real-time bilateral teleoperation setup. I am looking to read external forces acting on the panda's end-effector.
So far I can see that there are following topics available: F_ext, O_F_ext_hat_K, K_F_ext_hat_K, F_T_EE, .F_x_Cee, .F_x_Cload.
I would like to get some clarity on what exactly these forces are.

bomarali, 2019-03-22:
Found it here
https://frankaemika.github.io/libfranka/structfranka_1_1RobotState.html

bomarali, 2019-03-22:
a follow up question:
What is the stiffness frame?

aPonza, 2019-03-27:
I believe it's the tf frame [<arm_id>_K](https://frankaemika.github.io/docs/franka_ros.html#franka-control), which you can see in the tf tree, for example. It's published by the state controller in [here](https://github.com/frankaemika/franka_ros/blob/f7f00b6d9678e59e6f34ccc8d7aad6491b42ac80/franka_control/src/franka_state_controller.cpp#L404).

fwalch, 2019-07-22:
Closing due to inactivity. If you have more questions, feel free to re-open the issue.

closed 2019-07-22 by fwalch