frankarobotics/franka_ros #57: franka_control node die giving "Controller Spawner: Waiting for service controller_manager/load_controller"

artemiialessandrini, 2019-03-22:
Greetings!
So, while running franka_control.launch, topic dies with some errors.
Additionally I don't get any response from `rostopic pub -1 /franka_control/error_recovery/goal franka_control/ErrorRecoveryActionGoal "{}"`
While running `roslaunch franka_control franka_control.launch robot_ip:=172.16.0.2`, I am getting:
```
SUMMARY
========

PARAMETERS
 * /franka_control/arm_id: panda
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/robot_ip: 172.16.0.2
 * /franka_gripper/default_grasp_epsilon/inner: 0.005
 * /franka_gripper/default_grasp_epsilon/outer: 0.005
 * /franka_gripper/default_speed: 0.1
 * /franka_gripper/joint_names: ['panda_finger_jo...
 * /franka_gripper/publish_rate: 30
 * /franka_gripper/robot_ip: 172.16.0.2
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
 * /franka_state_controller/publish_rate: 30
 * /franka_state_controller/type: franka_control/Fr...
 * /joint_state_desired_publisher/rate: 30
 * /joint_state_desired_publisher/source_list: ['franka_state_co...
 * /joint_state_publisher/rate: 30
 * /joint_state_publisher/source_list: ['franka_state_co...
 * /position_joint_trajectory_controller/constraints/goal_time: 0.5
 * /position_joint_trajectory_controller/constraints/panda_joint1/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint2/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint3/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint4/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint5/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint6/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint7/goal: 0.05
 * /position_joint_trajectory_controller/joints: ['panda_joint1', ...
 * /position_joint_trajectory_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    franka_control (franka_control/franka_control_node)
    franka_gripper (franka_gripper/franka_gripper_node)
    joint_state_desired_publisher (joint_state_publisher/joint_state_publisher)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    state_controller_spawner (controller_manager/spawner)

auto-starting new master
process[master]: started with pid [10325]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 678a3aee-4c94-11e9-a975-4c72b9cc49b3
process[rosout-1]: started with pid [10338]
started core service [/rosout]
process[franka_gripper-2]: started with pid [10351]
process[franka_control-3]: started with pid [10356]
[ INFO] [1553253590.755850523]: franka_gripper_node: Found default_speed 0.1
process[state_controller_spawner-4]: started with pid [10358]
[ INFO] [1553253590.758443925]: franka_gripper_node: Found default_grasp_epsilon inner: 0.005, outer: 0.005
process[robot_state_publisher-5]: started with pid [10370]
process[joint_state_publisher-6]: started with pid [10386]
process[joint_state_desired_publisher-7]: started with pid [10391]
[INFO] [1553253591.413185]: Controller Spawner: Waiting for service controller_manager/load_controller
[WARN] [1553253621.717197]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[state_controller_spawner-4] process has finished cleanly
log file: /home/artemii/.ros/log/678a3aee-4c94-11e9-a975-4c72b9cc49b3/state_controller_spawner-4*.log
terminate called after throwing an instance of 'franka::NetworkException'
  what():  libfranka: Connection timeout
terminate called after throwing an instance of 'franka::NetworkException'
  what():  libfranka: Connection timeout
[franka_gripper-2] process has died [pid 10351, exit code -6, cmd /opt/ros/kinetic/lib/franka_gripper/franka_gripper_node __name:=franka_gripper __log:=/home/artemii/.ros/log/678a3aee-4c94-11e9-a975-4c72b9cc49b3/franka_gripper-2.log].
log file: /home/artemii/.ros/log/678a3aee-4c94-11e9-a975-4c72b9cc49b3/franka_gripper-2*.log
================================================================================REQUIRED process [franka_control-3] has died!
process has died [pid 10356, exit code -6, cmd /opt/ros/kinetic/lib/franka_control/franka_control_node __name:=franka_control __log:=/home/artemii/.ros/log/678a3aee-4c94-11e9-a975-4c72b9cc49b3/franka_control-3.log].
log file: /home/artemii/.ros/log/678a3aee-4c94-11e9-a975-4c72b9cc49b3/franka_control-3*.log
Initiating shutdown!
================================================================================
[joint_state_desired_publisher-7] killing on exit
[joint_state_publisher-6] killing on exit
[franka_control-3] killing on exit
[robot_state_publisher-5] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete

```
libfranka and franka-ros are installed as a deb:
```
ros-kinetic-libfranka is already the newest version (0.5.0-1xenial-20180810-152723-0800).
ros-kinetic-franka-ros is already the newest version (0.6.0-1xenial-20190220-185201-0800).

```
In other side, I do get a ping to the Franka's control and to the Franka base, while connected to it straighly.
`sudo ping 172.16.0.2 -i 0.001 -D -c 10000 -s 1200` gives me:

`rtt min/avg/max/mdev = 0.152/0.259/0.574/0.048 ms`,
which seems to be alright.

kernel version: `4.14.12-rt10`

cpufreq-info -p:
`1200000 3100000 performance`

I'm able to control Franka through the https://172.16.0.2/desk/, but not using ROS...

Looking forward to get some clue!
Thanks

bomarali, 2019-03-22:
Can you run the panda_moveit_config or any of the franka_example_controllers?

artemiialessandrini, 2019-03-22:
@bomarali, thank you for your response
no, nothing of those...
It also gets errors and eventually crashes

bomarali, 2019-03-22:
Reminds me of an error I get one I accidentally run the robot under the wrong kernel. Are you sure that you are on an rt kernel?
-uname -a

artemiialessandrini, 2019-03-22:
@bomarali, I guess, kernel is fine..
`Linux envy 4.14.12-rt10 #1 SMP PREEMPT RT Thu Mar 21 16:54:08 EDT 2019 x86_64 x86_64 x86_64 GNU/Linux`

artemiialessandrini, 2019-03-23:
Shall I add something in bashrc file?
e.g:
# export ROS_MASTER_URI=http://robot_ip:11311
# export ROS_IP=workspace_ip

bomarali, 2019-03-23:
Don't think you need that.

sgabl, 2019-03-25:
Please open Desk and check `Settings -> System -> Features`. Do you see there the `FCI (Franka Control Interface)` feature? This will be removed after a factory reset. If you don't have this feature file, please contact `support@franka.de` or your local supplier.

artemiialessandrini, 2019-03-27:
@sgabl , thank you!
You are right.

That's the actual problem, probably, someone run the factory reset and it got removed.

Solved.

closed 2019-07-05 by sgabl