frankarobotics/franka_ros #63: (Question) Franka_ros and CUDA

bomarali, 2019-05-09:
Hello,

This is more of a question rather than issue. I am having a blast fooling around with franka_ros. I want to run some RGBD sensing parallel to robot on CUDA. The FCI documentation states that PREEMPT_RT kernels are incompatible with NVIDIA drivers. I assume that means that running franka_ros and anything on gpu is a no-no?
So my question is: have anyone have a workaround for this (aside having another PC for cuda)?

thatrobotguy, 2019-05-30:
There was a post about this on the franka community forums, but this is the best option offered thus far by a forum member:

I found this guy here has gotten a real time kernel with NVidia drivers (for “gaming workloads”). I got it from a blog somewhere.
Source: https://liquorix.net

fwalch, 2019-07-22:
Closing this issue as there is a community post about this topic: https://www.franka-community.de/t/how-to-install-nvidia-driver-with-real-time-kernel/754/3

For completeness: in libfranka it is also possible to create a `franka::Robot` instance which does not enforce a realtime kernel by passing `RealtimeConfig::kIgnore` to the [constructor](https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#ae63bc19390df3d54f3a270814df35eb6). You can try how this works on your PC, but it will probably increase the number of FCI communication problems you will encounter.

closed 2019-07-22 by fwalch