frankarobotics/franka_ros #65: Publisher for Error.msg

akashcastelino, 2019-05-15:
Hi,

I can see that Franka_msg package of Franka_Ros has a message “Error.msg”.

But I wasnt able to find which topic publishes this message.

My objective is to get a notifcation when the robot goes into error and error_recovery is required. Could someone who knows which topic publishes the “error.msg” point out which topic it is ?

MattiaRacca, 2019-05-21:
Hi,

I was trying to achieve something similar and the topic you need to look at is `/franka_state_controller/franka_states` of type `FrankaState.msg`, that has 2 `Error.msg` embedded (definition of both is in `franka_msgs`).

However, I found that some errors are reported correctly in the `current_errors` field (e.g. `joint_position_limits_violation`) but others seem not reported (e.g. I never seen a `True` for `cartesian_reflex` in `current_errors`).

If the Franka team could tell us how the different errors are handled and published it would be great!

akashcastelino, 2019-05-22:
Hi @MattiaRacca ,

I found that in the FrankaState message , there is a field called 'robot mode' which becomes 4 when the robot is in error.

closed 2019-05-22 by akashcastelino