frankarobotics/franka_ros #66: gripper_node died because of Incompatible library version (server version: 2, library version: 3).

nkuzqy, 2019-05-26:
when i run 
roslaunch franka_gripper franka_gripper.launch robot_ip:=172.16.0.2 
> SUMMARY
========

PARAMETERS
 * /franka_gripper/default_grasp_epsilon/inner: 0.005
 * /franka_gripper/default_grasp_epsilon/outer: 0.005
 * /franka_gripper/default_speed: 0.1
 * /franka_gripper/joint_names: ['panda_finger_jo...
 * /franka_gripper/publish_rate: 30
 * /franka_gripper/robot_ip: 172.16.0.2
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    franka_gripper (franka_gripper/franka_gripper_node)

auto-starting new master
process[master]: started with pid [1681]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 6f82703c-7f83-11e9-8f2f-c8f75064a334
process[rosout-1]: started with pid [1694]
started core service [/rosout]
process[franka_gripper-2]: started with pid [1697]
[ INFO] [1558853810.733660038]: franka_gripper_node: Found default_speed 0.1
[ INFO] [1558853810.734204669]: franka_gripper_node: Found default_grasp_epsilon inner: 0.005, outer: 0.005
terminate called after throwing an instance of 'franka::IncompatibleVersionException'
  what():  libfranka: Incompatible library version (server version: 2, library version: 3). Please check https://frankaemika.github.io for Panda system updates or use a different version of libfranka.
[franka_gripper-2] process has died [pid 1697, exit code -6, cmd /home/dell/zs_franka_ws/devel/lib/franka_gripper/franka_gripper_node __name:=franka_gripper __log:=/home/dell/.ros/log/6f82703c-7f83-11e9-8f2f-c8f75064a334/franka_gripper-2.log].
log file: /home/dell/.ros/log/6f82703c-7f83-11e9-8f2f-c8f75064a334/franka_gripper-2*.log
^C[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

sgabl, 2019-07-05:
Please check http://support.franka.de/ to get the right version of `libfranka` and `franka_ros` for your robot image version (you'll see the robot image version in `Desk -> Settings -> Dashboard`)

closed 2019-07-05 by sgabl

a-z-e-r-i-l-a, 2020-08-18:
I checked robot image version in Dashboard which says 3.0.2, however I didn't find anything in support.franka.de, can you please guide me about it?

ninafiona, 2021-03-01:
I also get the error `libfranka: Incompatible library version (server version: 3, library version: 4)` and my server version is 3.0.2, which libfranka and franka_ros versions are compatible? I tried 0.7.1 (github tag).