frankarobotics/franka_ros #67: failed to start move_to_start.launch

lucafei, 2019-05-26:
hi,guys i have a problem at starting move_to_start.launch. the following are the issue codes
roslaunch franka_example_controllers move_to_start.launch robot_ip:=192.168.10.11
... logging to /home/yin/.ros/log/7b181bfc-7fa3-11e9-ac31-acd1b82882db/roslaunch-yin-ThinkPad-E555-12880.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://yin-ThinkPad-E555:41705/

SUMMARY
========

PARAMETERS
 * /franka_control/arm_id: panda
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/robot_ip: 192.168.10.11
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
 * /franka_state_controller/publish_rate: 30
 * /franka_state_controller/type: franka_control/Fr...
 * /joint_state_desired_publisher/rate: 30
 * /joint_state_desired_publisher/source_list: ['franka_state_co...
 * /joint_state_publisher/rate: 30
 * /joint_state_publisher/source_list: ['franka_state_co...
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'default': True...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.05
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/FMTkConfigDefault/heuristics: 0
 * /move_group/planner_configs/FMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/FMTkConfigDefault/num_samples: 1000
[... 246 lines cut ...]
[ERROR] [1558867601.543452891]: Apparently trajectory initialization failed
[ INFO] [1558867601.544476203]: ABORTED: Solution found but controller failed during execution
================================================================================REQUIRED process [move_to_start-9] has died!
process has finished cleanly
log file: /home/yin/.ros/log/7b181bfc-7fa3-11e9-ac31-acd1b82882db/move_to_start-9*.log
Initiating shutdown!
================================================================================
[move_to_start-9] killing on exit
[move_group-8] killing on exit
[controller_spawner-7] killing on exit
[state_controller_spawner-3] killing on exit
[joint_state_publisher-5] killing on exit
[robot_state_publisher-4] killing on exit
[joint_state_desired_publisher-6] killing on exit
[franka_control-2] killing on exit
[WARN] [1558867603.049634]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1558867603.077656]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[franka_control-2] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

i don't know why the node killed , so anybody have ever met this problem and solved this. pls help me

aPonza, 2019-05-30:
> Failed to import pyassimp, see ros-planning/moveit#86 for more info

Check the output of `pip freeze | grep pyassimp` in your terminal. You probably need to update to pyassimp >=3.3 like the link says.

After that, I had a similar issue, the solution seemed to be [this](https://github.com/frankaemika/franka_ros/issues/23#issuecomment-410169277) iirc, but do take a look at the exchange up to that point, tl;dr being build libfranka from source, update the firmware of your controller via Desk and https://frankaemika.github.io/

fwalch, 2019-07-22:
Closing due to inactivity. If you are still having this problem, feel free to re-open the issue.

closed 2019-07-22 by fwalch