frankarobotics/franka_ros #68: Unwanted automatic restart of current controller on ErrorRecoveryAction

konsiritt, 2019-05-27:
Hi,
I am performing a franka_control::ErrorRecoveryAction whenever I run into a collision that results in a "cartesian reflex". I would want to react to this event by switching to another custom controller, that resets the robot to initial state without contact. However, the default behavior of the ErrorRecoveryAction is to restart the currently running controller. Due to the time it takes my controller manager to switch controllers the automatic restart of the current controller leads to running into the collision repeatedly which I would want to avoid of course. 
Is there any way that this default behavior of restarting the current controller can be changed/optional?
Cheers for any help!

thatrobotguy, 2019-05-30:
Somebody mentioned on the forums that you may want to try implementing a try/catch statement for some franka errors, or at least handle the error before it gets pushed higher in the code

rhaschke, 2022-08-23:
I think switching the controller _before_ triggering error recovery should do the job.
This will actually start the newly selected controller.