frankarobotics/franka_ros #69: can't recover from joint limit error

hironoriyh, 2019-05-30:
Hi all,

 I can only recover from joint limit error via Desk and lock/unlock joints. 
Do you know how to recover from the error via publishing rostopic?

aPonza, 2019-05-30:
```
rostopic pub -1 /franka_control/error_recovery/goal franka_control/ErrorRecoveryActionGoal "{}"
```

from [here](https://frankaemika.github.io/docs/franka_ros.html).

hironoriyh, 2019-05-30:
I've tried publishing the message and could recover from most of errors except for joint limit error.

aPonza, 2019-05-30:
Weird, it does take a while sometimes, but it always works for me. Are you sure it's not some other condition? Can you test if with only `franka_control` up and the robot in a `joint_position_limits_violation` with the EAD activated, publishing the message doesn't work?

sgabl, 2019-07-22:
If this problem still persist, try publishing the topic to the `error_recovery` goal, download directly afterwards the logs of the robot and send it to `support@franka.de`.

closed 2019-07-22 by sgabl