frankarobotics/franka_ros #70: franka_control dies

GunnarMaer, 2019-06-20:
Hello everyone,

I try to control the Panda robot with ROS Melodic.  I tried a lot of things in simulation and now I try to work with the real robot. 

When trying to run specific parts of franka_ros, I always get errors that like 
`REQUIRED process [franka_control-1] has died!`

I tried to run all packages from [franka_ros]( https://frankaemika.github.io/docs/franka_ros.html) , but only the HW test (franka_hw franka_hw_test.test) and the visualization (franka_visualization franka_visualization.launch) works and looks fine. 

My OS is Ubuntu 18.04.2 LTS, running on a 
Intel® Xeon(R) CPU E3-1225 v3 @ 3.20GHz × 4. 

I already looked into other issues and these are the things that work or possible error sources I can exclude:

**Ping Test:**
```
[…]
--- 172.16.0.2 ping statistics ---
10000 packets transmitted, 10000 received, 0% packet loss, time 9998ms
rtt min/avg/max/mdev = 0.122/0.162/0.307/0.018 ms
```

**Echo Robot State works** (though not well formatted):
```
panda@boern-8:~$ cd libfranka/build/
panda@boern-8:~/libfranka/build$ ./examples/echo_robot_state
Usage: ./examples/echo_robot_state <robot-hostname>
panda@boern-8:~/libfranka/build$ ./examples/echo_robot_state 172.16.0.2
{"O_T_EE": [-0.964776,0.0863813,0.248487,0,0.243837,-0.0609248,0.967901,0,0.0987475,0.994398,0.0377158,0,0.0751279,0.0649383,1.13025,1], "O_T_EE_d": [1,0,0,0,0,-1,0,0,0,0,-1,0,0.088,0,0.926,1], "F_T_EE": [1,0,0,0,0,1,0,0,0,0,1,0,0,0,0,1], "EE_T_K": [1,0,0,0,0,1,0,0,0,0,1,0,0,0,0,1], "m_ee": 0.7, "F_x_Cee": [0,0,0], "I_ee": [0.001,0,0,0,0.001,0,0,0,0.001], "m_load": 0, "F_x_Cload": [0,0,0], "I_load": [0,0,0,0,0,0,0,0,0], "m_total": 0.7, "F_x_Ctotal": [0,0,0], "I_total": [0.001,0,0,0,0.001,0,0,0,0.001], "elbow": [-2.16897,1], "elbow_d": [0,1], "elbow_c": [0,0], "delbow_c": [0,0], "ddelbow_c": [0,0], "tau_J": [0.230636,2.91182,0.175241,-3.26469,0.0818798,0.762344,0.095393], "tau_J_d": [0,0,0,0,0,0,0], "dtau_J": [1.2411,51.2059,46.5436,-152.684,4.6843,-21.8679,-48.8285], "q": [1.85475,-0.0321982,-2.16897,-0.126436,1.7856,1.55162,-1.45504], "dq": [-0.000664768,-0.00086661,-0.000551295,0.000435586,-6.38721e-06,-0.000130034,0.000484985], "q_d": [0,0,0,0,0,0,0], "dq_d": [0,0,0,0,0,0,0], "ddq_d": [0,0,0,0,0,0,0], "joint_contact": [0,0,0,0,0,0,0], "cartesian_contact": [0,0,0,0,0,0], "joint_collision": [0,0,0,0,0,0,0], "cartesian_collision": [0,0,0,0,0,0], "tau_ext_hat_filtered": [0.213561,0.148066,0.0320331,0.00828305,-0.0541777,-0.541861,0.0663466], "O_F_ext_hat_K": [0,0,0,0,0,0], "K_F_ext_hat_K": [0,0,0,0,0,0], "O_dP_EE_d": [0,0,0,0,0,0], "O_T_EE_c": [0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0], "O_dP_EE_c": [0,0,0,0,0,0], "O_ddP_EE_c": [0,0,0,0,0,0], "theta": [1.85477,-0.0319931,-2.16895,-0.126665,1.78561,1.55171,-1.45503], "dtheta": [0,0,0,0,0,0,0], "current_errors": [], "last_motion_errors": [], "control_command_success_rate": 0, "robot_mode": "Other", "time": 18351086}
```

My systems runs on a realtime Kernel: 
```
panda@boern-8:~$ uname -a
Linux boern-8 4.19.37-rt20-rc1 #1 SMP PREEMPT RT Wed Jun 5 15:10:01 CEST 2019 x86_64 x86_64 x86_64 GNU/Linux
```

`ros-melodic-ros-control` and `ros-melodic-controller-manager` are installed.

Here is one example of the outputs and errors I'm getting:
```
panda@boern-8:~$ roslaunch franka_example_controllers move_to_start.launch robot_ip:=172.16.0.2 load_gripper:=false
... logging to /home/panda/.ros/log/e0bced24-935e-11e9-bb2e-f4f26d060b53/roslaunch-boern-8-13578.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
 
WARNING: Package name "configV3" does not follow the naming conventions. It should start with a lower case letter and only contain lower case letters, digits, underscores, and dashes.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://boern-8:37833/
 
SUMMARY
========
 
PARAMETERS
* /franka_control/arm_id: panda
* /franka_control/cutoff_frequency: 100
* /franka_control/internal_controller: joint_impedance
* /franka_control/joint_names: ['panda_joint1', ...
* /franka_control/rate_limiting: True
* /franka_control/robot_ip: 172.16.0.2
* /franka_state_controller/arm_id: panda
* /franka_state_controller/joint_names: ['panda_joint1', ...
* /franka_state_controller/publish_rate: 30
* /franka_state_controller/type: franka_control/Fr...
* /joint_state_desired_publisher/rate: 30
* /joint_state_desired_publisher/source_list: ['franka_state_co...
* /joint_state_publisher/rate: 30
* /joint_state_publisher/source_list: ['franka_state_co...
[... 332 lines cut ...]
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...
 
You can start planning now!
 
shutting down processing monitor...
... shutting down processing monitor complete
done

```

Depending on the package I run, I see a window (probably RViz) popping up and disappearing in the same second, while that output is created.

The configV3 should not be related to my problem. It is a package related to the MoveIt Setup Assistant Wizard. I replaced it with another name but had difficulties removing it completely. 

I saw this idea, but I don't find "features" under settings. Maybe I'm in the wrong settings? What exactly is meant with "Desk"?
> Please open Desk and check `Settings -> System -> Features`. Do you see there the `FCI (Franka Control Interface)` feature? This will be removed after a factory reset. If you don't have this feature file, please contact `support@franka.de` or your local supplier. (Similar to issue [38](https://github.com/frankaemika/libfranka/issues/38))

_Originally posted by @sgabl in https://github.com/frankaemika/franka_ros/issues/57#issuecomment-476168902_

I hope, you can help me!

Thank you very much!

sgabl, 2019-06-21:
Hi,

thanks for the detailed description!
As you get the robot state, the FCI feature is installed. You can access the Desk programming environment in your setup by navigating to `https://172.16.0.2/`

Having a look at the `robot_mode` (output of `./examples/echo_robot_state 172.16.0.2`) and `what():  libfranka: Set Collision Behavior command rejected: command not possible in the current mode!` I assume that the `fail-safe safety locking system` is not opened or the `external activation device` is not toggled / opened. You can open the `fail-safe safety locking system` in the Desk environment.

GunnarMaer, 2019-06-21:
Thanks a lot! 

This resolved my issue.

closed 2019-06-21 by GunnarMaer