frankarobotics/franka_ros #80: when i use the code "roslaunch franka_example_controllers joint_impedance_example_controller.launch robot_ip:=172.16.0.2" some wrong happend .

trojantao, 2019-11-02:
i use the melodic in ubuntu18.04
$ roslaunch franka_example_controllers joint_impedance_example_controller.launch robot_ip:=172.16.0.2
... logging to /home/trojan/.ros/log/4bd7545c-fd39-11e9-ae61-8c16458430ed/roslaunch-Y7000-11000.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://Y7000:46121/

SUMMARY
========

PARAMETERS
 * /cartesian_impedance_example_controller/arm_id: panda
 * /cartesian_impedance_example_controller/joint_names: ['panda_joint1', ...
 * /cartesian_impedance_example_controller/type: franka_example_co...
 * /cartesian_pose_example_controller/arm_id: panda
 * /cartesian_pose_example_controller/type: franka_example_co...
 * /cartesian_velocity_example_controller/arm_id: panda
 * /cartesian_velocity_example_controller/type: franka_example_co...
 * /elbow_example_controller/arm_id: panda
 * /elbow_example_controller/type: franka_example_co...
 * /force_example_controller/arm_id: panda
 * /force_example_controller/joint_names: ['panda_joint1', ...
 * /force_example_controller/type: franka_example_co...
 * /franka_control/arm_id: panda
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/robot_ip: 172.16.0.2
 * /franka_gripper/default_grasp_epsilon/inner: 0.005
 * /franka_gripper/default_grasp_epsilon/outer: 0.005
 * /franka_gripper/default_speed: 0.1
 * /franka_gripper/joint_names: ['panda_finger_jo...
 * /franka_gripper/publish_rate: 30
 * /franka_gripper/robot_ip: 172.16.0.2
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
 * /franka_state_controller/publish_rate: 30
 * /franka_state_controller/type: franka_control/Fr...
 * /joint_impedance_example_controller/acceleration_time: 2.0
 * /joint_impedance_example_controller/arm_id: panda
 * /joint_impedance_example_controller/coriolis_factor: 1.0
 * /joint_impedance_example_controller/d_gains: [50.0, 50.0, 50.0...
 * /joint_impedance_example_controller/joint_names: ['panda_joint1', ...
 * /joint_impedance_example_controller/k_gains: [600.0, 600.0, 60...
 * /joint_impedance_example_controller/publish_rate: 10.0
 * /joint_impedance_example_controller/radius: 0.1
 * /joint_impedance_example_controller/type: franka_example_co...
 * /joint_impedance_example_controller/vel_max: 0.15
 * /joint_position_example_controller/joint_names: ['panda_joint1', ...
 * /joint_position_example_controller/type: franka_example_co...
 * /joint_state_desired_publisher/rate: 30
 * /joint_state_desired_publisher/source_list: ['franka_state_co...
 * /joint_state_publisher/rate: 30
 * /joint_state_publisher/source_list: ['franka_state_co...
 * /joint_velocity_example_controller/joint_names: ['panda_joint1', ...
 * /joint_velocity_example_controller/type: franka_example_co...
 * /model_example_controller/arm_id: panda
 * /model_example_controller/type: franka_example_co...
 * /position_joint_trajectory_controller/constraints/goal_time: 0.5
 * /position_joint_trajectory_controller/constraints/panda_joint1/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint2/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint3/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint4/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint5/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint6/goal: 0.05
 * /position_joint_trajectory_controller/constraints/panda_joint7/goal: 0.05
 * /position_joint_trajectory_controller/joints: ['panda_joint1', ...
 * /position_joint_trajectory_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3

NODES
  /
    controller_spawner (controller_manager/spawner)
    franka_control (franka_control/franka_control_node)
    franka_gripper (franka_gripper/franka_gripper_node)
    joint_state_desired_publisher (joint_state_publisher/joint_state_publisher)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    state_controller_spawner (controller_manager/spawner)

ROS_MASTER_URI=http://localhost:11311

process[franka_gripper-1]: started with pid [11018]
process[franka_control-2]: started with pid [11019]
[ INFO] [1572676841.063536417]: franka_gripper_node: Found default_speed 0.1
[ INFO] [1572676841.064317797]: franka_gripper_node: Found default_grasp_epsilon inner: 0.005, outer: 0.005
process[state_controller_spawner-3]: started with pid [11024]
process[robot_state_publisher-4]: started with pid [11032]
process[joint_state_publisher-5]: started with pid [11037]
process[joint_state_desired_publisher-6]: started with pid [11040]
process[controller_spawner-7]: started with pid [11044]
process[rviz-8]: started with pid [11046]
[ INFO] [1572676841.212762359]: rviz version 1.13.3
[ INFO] [1572676841.212812619]: compiled against Qt version 5.9.5
[ INFO] [1572676841.212826286]: compiled against OGRE version 1.9.0 (Ghadamon)
[INFO] [1572676841.393049]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1572676841.394721]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1572676841.396224]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1572676841.397775]: Loading controller: joint_impedance_example_controller
[INFO] [1572676841.405241]: Controller Spawner: Loaded controllers: joint_impedance_example_controller
[ INFO] [1572676841.407720128]: FrankaHW: Prepared switching controllers to joint_torque, cartesian_pose with parameters limit_rate=1, cutoff_frequency=100, internal_controller=joint_impedance
terminate called after throwing an instance of 'franka::RealtimeException'
  what():  libfranka: Running kernel does not have realtime capabilities.
[INFO] [1572676841.409264]: Started controllers: joint_impedance_example_controller
[INFO] [1572676841.431992]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1572676841.433632]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1572676841.434862]: Controller Spawner: Waiting for service controller_manager/unload_controller
**_**```
[INFO] [1572676841.435992]: Loading controller: franka_state_controller
Traceback (most recent call last):
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 207, in <module>
    if __name__ == '__main__': main()
  File "/opt/ros/melodic/lib/controller_manager/spawner", line 185, in main
    resp = load_controller(name)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 439, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/rospy/impl/tcpros_service.py", line 509, in call
    raise ServiceException("unable to connect to service: %s"%e)
rospy.service.ServiceException: unable to connect to service: [Errno 104] Connection reset by peer
[INFO] [1572676841.547714]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1572676841.548362]: Stopping all controllers...
[WARN] [1572676841.549811]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
================================================================================REQUIRED process [franka_control-2] has died!
process has died [pid 11019, exit code -6, cmd /home/trojan/catkin_ws/devel/lib/franka_control/franka_control_node __name:=franka_control __log:=/home/trojan/.ros/log/4bd7545c-fd39-11e9-ae61-8c16458430ed/franka_control-2.log].
log file: /home/trojan/.ros/log/4bd7545c-fd39-11e9-ae61-8c16458430ed/franka_control-2*.log
Initiating shutdown!
================================================================================
```**_**
[rviz-8] killing on exit
[controller_spawner-7] killing on exit
[joint_state_desired_publisher-6] killing on exit
[joint_state_publisher-5] killing on exit
[robot_state_publisher-4] killing on exit
[state_controller_spawner-3] killing on exit
[franka_gripper-1] killing on exit
[INFO] [1572676841.693296]: Shutting down spawner. Stopping and unloading controllers...
[franka_control-2] killing on exit
[INFO] [1572676841.694207]: Stopping all controllers...
[WARN] [1572676841.696933]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 111] Connection refused
shutting down processing monitor...
... shutting down processing monitor complete
done

fwalch, 2019-11-06:
The issue is pointed out in the output:

```
 libfranka: Running kernel does not have realtime capabilities.
```

Setting up a realtime kernel is described at https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel

closed 2019-11-06 by fwalch

FrankJIE09, 2021-08-11:
> The issue is pointed out in the output:
> 
> ```
>  libfranka: Running kernel does not have realtime capabilities.
> ```
> 
> Setting up a realtime kernel is described at https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel

Thanks.You help me solve the same issue.