frankarobotics/franka_ros #81: Can't install libfranka and franka-ros

christophecricket, 2019-11-05:
When I try to install these through the ROS repositories with ROS melodic already installed, I get an error message that apt can't find the .deb files it's looking for, and indeed, the .deb files it's looking for don't match the ones on the ROS repo. What am I doing wrong? My ROS melodic install is fully up to date and I simply tried: `sudo apt install ros-melodic-libfranka ros-melodic-franka-ros`

fwalch, 2019-11-06:
Did you try running `sudo apt-get update` before? Otherwise, can you post the full error output?

christophecricket, 2019-11-06:
Funnily enough I did not. Sorry, I'm used to the more AUR-style system of package management. Thanks for humoring me. Everything works.

closed 2019-11-06 by christophecricket