frankarobotics/franka_ros #82: Load a streaming controller by default

AndyZe, 2019-11-06:
As I understand it, only a trajectory controller is loaded by default. It would be great if a JointGroupPositionController or JointGroupVelocityController was also loaded so we can stream real-time commands to the robot.

This type of command is pretty popular with UR robots. Example:

https://drive.google.com/file/d/1i77SVG8YFoIGYAzKRylypvYj8J8b838f/view

R34ls0cr4t3, 2023-06-13:
import requests
url = "http://123.123.123.123/move"
params_position1 = {
    "Joint1": 1,
    "Joint2": 2,
    "Joint3": 3,
    "Joint4": 4,
    "Joint5": 5,
    "Joint6": 6,
    "Joint7": 7
}
params_position2 = {
    "Joint1": 1,
    "Joint2": 2,
    "Joint3": 3,
    "Joint4": 4,
    "Joint5": 5,
    "Joint6": 6,
    "Joint7": 7
}
for i in range(5):
    response = requests.get(url, params=params_position2, verify=False)
    response = requests.get(url, params=params_position1, verify=False)

fe-support, 2023-06-20:
Closing due to inactivity. Please open this git issue again if the topic still exists.

closed 2023-06-20 by fe-support