frankarobotics/franka_ros #83: Could not load controller 'position_joint_trajectory_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.

trojantao, 2019-11-09:
roslaunch franka_example_controllers move_to_start.launch robot_ip:=172.16.0.2
... logging to /home/trojan/.ros/log/4cb413c0-029c-11ea-9ce4-8c16458430ed/roslaunch-Y7000-4875.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Y7000:41443/

SUMMARY
========

PARAMETERS
 * /franka_control/arm_id: panda
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/robot_ip: 172.16.0.2
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
 * /franka_state_controller/publish_rate: 30
 * /franka_state_controller/type: franka_control/Fr...
 * /joint_state_desired_publisher/rate: 30
 * /joint_state_desired_publisher/source_list: ['franka_state_co...
 * /joint_state_publisher/rate: 30
 * /joint_state_publisher/source_list: ['franka_state_co...
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'default': True...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.05
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/FMTkConfigDefault/heuristics: 0
 * /move_group/planner_configs/FMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/FMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/FMTkConfigDefault/radius_multiplier: 1.1
[... 259 lines cut ...]
[ INFO] [1573268120.246729112]: ABORTED: Solution found but controller failed during execution
================================================================================REQUIRED process [move_to_start-9] has died!
process has finished cleanly
log file: /home/trojan/.ros/log/4cb413c0-029c-11ea-9ce4-8c16458430ed/move_to_start-9*.log
Initiating shutdown!
================================================================================
[move_to_start-9] killing on exit
[move_group-8] killing on exit
[joint_state_desired_publisher-6] killing on exit
[controller_spawner-7] killing on exit
[joint_state_publisher-5] killing on exit
[robot_state_publisher-4] killing on exit
[state_controller_spawner-3] killing on exit
[franka_control-2] killing on exit
[INFO] [1573268121.357164]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1573268121.358378]: Stopping all controllers...
[INFO] [1573268121.364873]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1573268121.366201]: Stopping all controllers...
[WARN] [1573268121.422813]: Controller Spawner error while taking down controllers: transport error completing service call: unable to receive data from sender, check sender's logs for details
[WARN] [1573268121.512119]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

trojantao, 2019-11-09:
when i want to use the demo flaunch  move_to_start.launch meet the error

trannguyenle95, 2019-11-25:
I think the error occurs because you don't have JointTrajectoryController. The same problem and solution are addressed [here]( https://answers.ros.org/question/254084/gazebo-could-not-load-controller-jointtrajectorycontroller-does-not-exist-mastering-ros-chapter-10/).

gollth, 2022-01-03:
Hi @trojantao it looks like you have no controller running, which can execute a position trajectory:

```
[ INFO] [1573268120.245273572]: Returned 0 controllers in list
[ERROR] [1573268120.245444769]: Unable to identify any set of controllers that can actuate the specified joints: [ panda_joint1 panda_joint2 panda_joint3 panda_joint4 panda_joint5 panda_joint6 panda_joint7 ]
```

Like @trannguyenle95 suggested, installing the controller might help in your setup. I'll close this ticket, but feel free to reopen if the issue persists.

P.S: Please format your logs in code blocks next time for better readability

closed 2022-01-03 by gollth