frankarobotics/franka_ros #87: Getting FrankaModelInterface in franka_state_controller.cpp

danfeiX, 2019-12-18:
I'm trying to publish jacobian from the `franka_state_controller` node. To do that, I need to get a `FrankaModelInterface` from the `robot_hardware` object passed into the `init()` function defined in `franka_state_controller.cpp`. I tried to add this line

```
auto* model_interface = robot_hardware->get<franka_hw::FrankaModelInterface>();
```

However, this function call always returns `nullptr`. Any idea what might be the cause?

Thanks

CarolinePascal, 2021-02-08:
Hi ! 
This post dates back a bit, but I was wondering : Did you find a way to publish the jacobian matrix ?
Thanks for you answer !

rtkg, 2021-06-24:
I think you might need to add `franka_hw::FrankaModelInterface` as template parameter to he MultiInterfaceController when declaring the FrankaStateController class.

gollth, 2021-12-06:
Like @rtkg suggested, that will the most likely fix of your problem. Feel free to reopen if the issue persists

closed 2021-12-06 by gollth