frankarobotics/franka_ros #91: Solidworks CAD model of Franka Emika Panda arm.

pkr97, 2020-01-22:
I would like to generate URDF from Solidworks and for that, I must have Solidworks CAD model.
I want to learn from the basics. For that, I need simple cad models of Franka Emika Panda arm. I want to assemble them in Solidworks and then I can generate the corresponding URDF.

I see that there are cad files in .STL and .obj format but I'm unable to use them in SolidWorks for assembling.

Any help is appreciated.

Thanks.

cheiio, 2021-06-07:
Hello pkr97,
did you get a SW assembly for this ?

pkr97, 2021-06-07:
Hey cheiio, not yet. I also stopped looking for it. But downloaded the .stl files and then used it with MoveIt and it was working fine.

gollth, 2021-12-27:
Hi @pkr97,

sorry for the late reply.

May I ask, what you want to do with such as assembly? You surely understand that many of the internals of the robot's design are proprietary and confidential. 

If it's just about reassembling the kinematic structure of the robot you can easily reconstruct this from the [DH-parameters](https://frankaemika.github.io/docs/control_parameters.html#denavithartenberg-parameters) and the [visual models](https://github.com/frankaemika/franka_ros/tree/develop/franka_description/meshes/visual).

I'll close this ticket now, since it's probably solved by now, but don't hesitate to reopen

closed 2021-12-27 by gollth