frankarobotics/franka_ros #96: errors when launching franka_example_controllers in latest franka_ros version

kmerckae, 2020-02-25:
Hello everyone,

I already tried launching several franka_example_controllers, but I always get some errors because of which the examples don't work. 

For example, when I run roslaunch franka_example_controllers move_to_start.launch robot_ip:=192.168.2.106 load_gripper:=true, I get 
```
... logging to /home/kmerckae/.ros/log/a1eb4d20-57a8-11ea-9522-7085c2dbe2ca/roslaunch-franka-desktop-19604.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://franka-desktop:43801/

SUMMARY
========

PARAMETERS
 * /franka_control/arm_id: panda
 * /franka_control/collision_config/lower_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/lower_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_limit_warning_threshold: 0.1
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/realtime_config: enforce
 * /franka_control/robot_ip: 192.168.2.106
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
 * /franka_state_controller/publish_rate: 30
 * /franka_state_controller/type: franka_control/Fr...
 * /joint_state_desired_publisher/rate: 30
 * /joint_state_desired_publisher/source_list: ['franka_state_co...
 * /joint_state_publisher/rate: 30
 * /joint_state_publisher/source_list: ['franka_state_co...
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'default': True...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.05
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
[... 274 lines cut ...]
[controller_spawner-7] killing on exit
[joint_state_desired_publisher-6] killing on exit
[joint_state_publisher-5] killing on exit
[robot_state_publisher-4] killing on exit
[INFO] [1582619241.540032]: Shutting down spawner. Stopping and unloading controllers...
[state_controller_spawner-3] killing on exit
[franka_control-2] killing on exit
[INFO] [1582619241.541110]: Stopping all controllers...
[INFO] [1582619241.542237]: Shutting down spawner. Stopping and unloading controllers...
[INFO] [1582619241.543892]: Stopping all controllers...
[WARN] [1582619241.611134]: Controller Spawner error while taking down controllers: unable to connect to service: connection from sender terminated before handshake header received. 0 bytes were received. Please check sender for additional details.
[WARN] [1582619241.611655]: Controller Spawner error while taking down controllers: unable to connect to service: [Errno 104] Connection reset by peer
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

The only way I could **solve this problem**, was by **going back to franka_ros release version 0.6.0** with git checkout 49e5ac1055e332581b4520a1bd9ac8aaf4580fb1, in which I can launch all the franka_example_controllers.

Can this problem be solved in the latest franka_ros version? 

Thanks a lot, 
Kelly

gollth, 2022-01-05:
> Found a contact between 'panda_link8' (type 'Robot link') and 'panda_link7' (type 'Robot link'), which constitutes a collision. Contact information is not stored.

Seems to be related to the MoveIT upgrade in https://github.com/frankaemika/franka_ros/issues/193 & https://github.com/ros-planning/panda_moveit_config/pull/66

samarth-robo, 2022-06-04:
@kmerckae a simple temporary workaround for franka_ros 0.9.0 is here: https://github.com/frankaemika/franka_ros/issues/243#issuecomment-1146700600.