frankarobotics/franka_ros #97: Self-Collision

Olrik1, 2020-02-28:
The Collision volumes of the robot are too big or misplaced.  
This leads to unwanted self-collisions.

One such example is link8 (see picture) which collides with link7 and a few others.  
The link is not set to adjacent in any of the srdf files.
An other interesting thing about this is, that the link did not use to have any collision mask before switching from models to the current shape based setups.  
Why is it needed now?

[image]

Further more, the length and position of link1 make it go through the base into anything that the robot is mounted on. When simulating a table in the planning scene the robot is continually in collision. (see picture)

[image]
 

It would be very nice if someone who is familiar with the model the robot could check the collision objects and redo them a bit more precisely wherever needed.

Thank you in advance!

blackchacal, 2020-03-07:
Hi @Olrik1.

I just updated franka_ros from melodic-devel branch. I realized that the collision behaviour changed. Before, there were STL collision files. Right now they just use geometries. For my purposes, I just used the DAE files used for visual representation as collision meshes. This allows for very precise collisions on rviz and gazebo. The only eventual problem is the absence of collision tolerances.

I do agree that Franka team should provide more precise collision geometry, either through STL files derived from the DAE files, the DAE files themselves, or through more precise geometry combination of cilinders, spheres, etc (which I think is not the best option).

Best regards.

pantor, 2020-05-13:
I can confirm that this is a major issue while updating to v0.7.0.