frankarobotics/franka_ros #100: Error when running franka_example_controller move_to_start.launch

Innoconnect, 2020-03-23:
Hi, I installed libfranka and frankaros from source, and I can successfully run the /libfranka/build/examples/communication_test, but when I run the move_to_start.launch, it never works, could you please help me? Thanks.
`silu@hrc:~/libfranka/build/examples$ roslaunch franka_example_controllers move_to_start.launch robot_ip:=192.1618.3.133
... logging to /home/silu/.ros/log/9a5cdcd8-6ccd-11ea-a21b-e454e8d91b88/roslaunch-hrc-23762.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://hrc:34725/

SUMMARY
========

PARAMETERS
 * /franka_control/arm_id: panda
 * /franka_control/collision_config/lower_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/lower_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/lower_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_force_thresholds_acceleration: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_force_thresholds_nominal: [20.0, 20.0, 20.0...
 * /franka_control/collision_config/upper_torque_thresholds_acceleration: [20.0, 20.0, 18.0...
 * /franka_control/collision_config/upper_torque_thresholds_nominal: [20.0, 20.0, 18.0...
 * /franka_control/cutoff_frequency: 100
 * /franka_control/internal_controller: joint_impedance
 * /franka_control/joint_limit_warning_threshold: 0.1
 * /franka_control/joint_names: ['panda_joint1', ...
 * /franka_control/rate_limiting: True
 * /franka_control/realtime_config: enforce
 * /franka_control/robot_ip: 192.1618.3.133
 * /franka_state_controller/arm_id: panda
 * /franka_state_controller/joint_names: ['panda_joint1', ...
 * /franka_state_controller/publish_rate: 30
 * /franka_state_controller/type: franka_control/Fr...
 * /joint_state_desired_publisher/rate: 30
 * /joint_state_desired_publisher/source_list: ['franka_state_co...
 * /joint_state_publisher/rate: 30
 * /joint_state_publisher/source_list: ['franka_state_co...
 * /move_group/allow_trajectory_execution: True
 * /move_group/controller_list: [{'default': True...
 * /move_group/hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_frame: camera_rgb_optica...
 * /move_group/octomap_resolution: 0.05
 * /move_group/panda_arm/planner_configs: ['SBLkConfigDefau...
 * /move_group/panda_arm_hand/planner_configs: ['SBLkConfigDefau...
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
[... 248 lines cut ...]
[ INFO] [1584944110.922815894]: No solution found after 0.000137 seconds
[ INFO] [1584944110.922985918]: Unable to solve the planning problem
[ INFO] [1584944110.923442040]: ABORTED: No motion plan found. No execution attempted.
================================================================================REQUIRED process [move_to_start-9] has died!
process has finished cleanly
log file: /home/silu/.ros/log/9a5cdcd8-6ccd-11ea-a21b-e454e8d91b88/move_to_start-9*.log
Initiating shutdown!
================================================================================
[move_to_start-9] killing on exit
[move_group-8] killing on exit
[controller_spawner-7] killing on exit
[joint_state_desired_publisher-6] killing on exit
[joint_state_publisher-5] killing on exit
[state_controller_spawner-3] killing on exit
[robot_state_publisher-4] killing on exit
[franka_control-2] killing on exit
[WARN] [1584944112.043737]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[WARN] [1584944112.044476]: Controller Spawner couldn't find the expected controller_manager ROS interface.
[franka_control-2] escalating to SIGTERM
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

Innoconnect, 2020-04-02:
I reinstalled all of the package again, panda_moveit_config, franka_ros, and moveit, afterwards the problem was solved.

closed 2020-04-02 by Innoconnect

Simon-Fuhaoyuan, 2020-08-14:
Hello! I also faced your problem, completely the same. However, after reinstalling all these packages again, I still cannot fix this bug. Can you please explain how you install these packages in detail? For example, you install these packages in binary or from source?