frankarobotics/franka_ros #101: Realtime kernel install fails with ''fakeroot: nested operation not yet supported"

felixvd, 2020-03-24:
The instructions on [this page](https://frankaemika.github.io/docs/installation_linux.html#setting-up-the-real-time-kernel) do not work for me on Ubuntu 18.04 trying to install kernel 5.2.21. I get this error message at the step `fakeroot make deb-pkg`:

```bash
make clean
/bin/bash ./scripts/package/mkdebian
  TAR     linux-5.2.21-rt15.tar.gz
origversion=$(dpkg-parsechangelog -SVersion |sed 's/-[^-]*$//');\
	mv linux-5.2.21-rt15.tar.gz ../linux-5.2.21-rt15_${origversion}.orig.tar.gz
dpkg-buildpackage -r"fakeroot -u" -a$(cat debian/arch)  -i.git -us -uc
dpkg-buildpackage: warning: using a gain-root-command while being root
dpkg-buildpackage: info: source package linux-5.2.21-rt15
dpkg-buildpackage: info: source version 5.2.21-rt15-1
dpkg-buildpackage: info: source distribution bionic
dpkg-buildpackage: info: source changed by root <root@osx-panda-desk>
dpkg-buildpackage: info: host architecture amd64
 dpkg-source -i.git --before-build TESTlin
 fakeroot -u debian/rules clean
fakeroot: FAKEROOTKEY set to 1483965928
fakeroot: nested operation not yet supported
dpkg-buildpackage: error: fakeroot -u debian/rules clean subprocess returned exit status 1
scripts/package/Makefile:71: recipe for target 'deb-pkg' failed
make[1]: *** [deb-pkg] Error 1
Makefile:1422: recipe for target 'deb-pkg' failed
make: *** [deb-pkg] Error 2
```

mahaarbo, 2021-01-25:
I'm not entirely sure, but could it be that you are still within the sudo session of the previous `apt-get install` command (assuming you ran that using sudo or something else previously with sudo the last 15 minutes)? 
Then fakeroot tries to set up a fake root while running as root which gives the nested operation error.

CodeFinder2, 2021-05-12:
Having the same problem and, no, not within a previous sudo session (explicitly used `sudo -k`).

Any ideas?

JGioia, 2021-05-15:
I had this issue when installing the real-time kernel. I am not sure what the problem is, but I ended up finding a different tutorial to install real-time kernel that worked. Both of these tutorials worked for me, however I could not start from where the franka tutorial stopped working (I had to start from the beginning): https://stackoverflow.com/questions/51669724/install-rt-linux-patch-for-ubuntu.
https://hmenn.github.io/pages/UbuntuRT_patch.html. For reference, the tutorial in the ROS 2 documentation also did not work for me.

rickstaa, 2021-05-19:
Related to https://github.com/frankaemika/libfranka/issues/62.

gollth, 2021-12-06:
We updated the documentation like @rickstaa mentioned. Now `fakeroot` shouldn't be needed anymore. Check the current documentation [here](https://frankaemika.github.io/docs/installation_linux.html)

closed 2021-12-06 by gollth