frankarobotics/franka_ros #104: Release into ROS Noetic?

davetcoleman, 2020-06-02:
ROS 1 Noetic is released and the MoveIt project is working on migrating to this new LTS (and last version of ROS 1)

So that the MoveIt tutorials with the Panda still work, can you all test your repo on Noetic and consider making a ``noetic-devel`` branch? Thanks!

davetcoleman, 2020-06-03:
In particular, we are seeing the following error, even when we install this package for source on Noetic:

> franka_gripper: Cannot locate rosdep definition for [libfranka]

rhaschke, 2020-06-29:
Actually, we only need franka_description to be released for MoveIt. Would be great, if somebody could handle this soon.

tylerjw, 2020-08-19:
One more :+1: for a Noetic release.  We need it released for us to be able to release MoveIt for Noetic.

davetcoleman, 2020-08-19:
I just emailed Sebastian and Florian at Franka

tylerjw, 2020-09-30:
Looks like this was done, and is now in our court so franka can release the rest of their packages.

davetcoleman, 2020-10-07:
@tylerjw i believe this particular issue can be closed now, correct?

tylerjw, 2020-10-07:
yes, someone with permissions on this repo has to close the issue.

closed 2020-10-14 by sgabl