frankarobotics/franka_ros #105: roslaunch franka_example_contollers joint_position_example_controller_sim.launch error

akshay-ka, 2020-06-22:
I get the following error when i'm  executing "roslaunch franka_example_contollers joint_position_example_controller_sim.launch"

ERROR:
RLException: [joint_position_example_controller_sim.launch] is neither a launch file in package [franka_example_contollers] nor is [franka_example_contollers] a launch file name
The traceback for the exception was written to the log file
-----------------------------------------------------------------------------------------------------------------------------------

Any solution for this error ??

peasant98, 2020-08-02:
Hi @akshay-ka, I looked through the launch files in https://github.com/frankaemika/franka_ros/tree/kinetic-devel/franka_example_controllers/launch. There seems to be no such launch file named `joint_position_example_controller_sim.launch`, which is why you're getting that error. Maybe you mean `joint_position_example_controller.launch`?

gollth, 2021-12-27:
@peasant98 thanks for the help. I'm also unaware of such a `*_sim.launch` file. @akshay-ka please note: if you want to simulate the robot, you can use the `franka_gazebo` package: https://frankaemika.github.io/docs/franka_ros.html#franka-gazebo

closed 2021-12-27 by gollth