frankarobotics/franka_ros #108: Grasp action always fails

bocsik, 2020-07-15:
I tried to write a simple program to see how the grasp action works:

    #include <ros/ros.h>
    #include <actionlib/client/simple_action_client.h>
    #include <actionlib/client/terminal_state.h>
    #include <franka_gripper/GraspAction.h>

    int main (int argc, char **argv)
    {
    ros::init(argc, argv, "gripper_test");

    actionlib::SimpleActionClient<franka_gripper::GraspAction> ac("franka_gripper/grasp", true);

    ROS_INFO("Waiting for action server to start.");

    ac.waitForServer();

   

    franka_gripper::GraspGoal goal;
    goal.width = 0.045;
    goal.speed = 0.03;
    goal.force = 20;

    ac.sendGoal(goal);

    bool finished_before_timeout = ac.waitForResult(ros::Duration(30.0));

    if (finished_before_timeout)
    {
      ROS_INFO("Action finished.");
    }
    else
      ROS_INFO("Action did not finish before the time out.");
    }

      return 0;
    }

When I try to grasp a 0.045m wide cube, the gripper lets go after about 1sec. I tried changing the epsilon values (even to very high values, like 0.08), but it didn't help.
When I use the Gripper::Grasp function from libranka with the same values I can grasp the cube without issue.

I'm using ROS Melodic, franka_ros 0.6.0 and server version 4.0.0 on Ubuntu 18.04.

Khesbak, 2020-08-05:
These are the necessary steps to grasp with a certain force an object.
I think that you did it all except the (goal.epsilon.inner  & goal.epsilon.outer) settings !!!!

    actionlib::SimpleActionClient<franka_gripper::GraspAction> ac("franka_gripper/grasp", true);
    goal.width = Grasp_Width;
    goal.speed = 0.1;           //  Closing speed. [m/s]
    goal.force = 60;            //   Grasping (continuous) force [N]
    goal.epsilon.inner = 0.05;  // Maximum tolerated deviation when the actual grasped width is
                                // smaller than the commanded grasp width.
    goal.epsilon.outer = 0.05;  // Maximum tolerated deviation when the actual grasped width is
                                // larger than the commanded grasp width.
    ac.sendGoal(goal);          // Sending the Grasp command to gripper

I added two intersting examples in the below repository, where a pick up object example exists.

    https://github.com/Khesbak/franka_ros

I hope this will help you

bocsik, 2020-08-07:
As far as I know epsilon inner and epsilon outer have a default value of 0.005, so the code should work even if they are not specified.

Khesbak, 2020-08-11:
Hi.
Actually, my experience was as I stated before, if you do not specify the epsilon values,  it will not grasp the object with the specified force and it will be released after one second.

gollth, 2021-12-27:
Hi @bocsik,

you are on a good track, just missing the `epsilon` settings like @Khesbak has suggested.

> As far as I know epsilon inner and epsilon outer have a default value of 0.005

Yes and no. There are actually two similar actions provided by `franka_gripper`, namely `grasp` and `gripper_command`. The first one is the more general action, which allows you to specify arbitrary epsilons, whereas the latter is used mainly by MoveIT and is a specialization of `grasp` with the "default_grasp_epsilon" set for both epsilons. See here for the relevant code:

https://github.com/frankaemika/franka_ros/blob/8efef6235b969c68ec1f8d5bc85026e4bca9cf19/franka_gripper/src/franka_gripper_node.cpp#L125-L131

So in your code above you are accessing the more general `grasp` action:

```c++
actionlib::SimpleActionClient<franka_gripper::GraspAction> ac("franka_gripper/grasp", true);
```

When you create your new goal, each member will be initialized with its _default_ value, which is 0.0 for the `epsilon.inner` & `epsilon.outer` doubles:

```c++
franka_gripper::GraspGoal goal;
```

## Solutions
1. Apply the fix from @Khesbak 
2. Use the `/franka_gripper/gripper_command` action instead of `/franka_gripper/grasp` and optionally provide another default than https://github.com/frankaemika/franka_ros/blob/8efef6235b969c68ec1f8d5bc85026e4bca9cf19/franka_gripper/config/franka_gripper_node.yaml#L3-L5

closed 2021-12-27 by gollth