frankarobotics/franka_ros #109: Integration with simulation in Gazebo

peasant98, 2020-07-28:
Hi all,

I was wondering if it would be possible sometime for integration with the Gazebo simulation.

https://github.com/erdalpekel/panda_simulation is gaining a lot of traction, and it would help to have this integrated (which I would be willing to help take on if allowed) into one central repository. The panda_simulation relies on forked versions as well, which makes things a little hectic.

Thanks!

gollth, 2021-08-05:
Hello and sorry for the late reply.

We are happy to announce that with version 0.8.0 of franka_ros there is Gazebo integration for the panda. Next to the necessary URDF changes (like adding estimated inertas) this includes also:

* Gravity Compensation in Gazebo, similar to how the real Panda does it
* Offering the same `FrankaState`- & `FrankaModelInterface` as `franka_control` does
* Offering the same action interfaces like `franka_gripper` does
* Offering similar service (non-realtime) interfaces similar to how `franka_control` does it

Hopefully this will allow you to simulate existing ROS controllers now in Gazebo as well.

Check out our [documentation](https://frankaemika.github.io/docs/franka_ros.html#franka-gazebo) for more information.

closed 2021-08-05 by gollth

peasant98, 2021-08-05:
Sweet, thanks!

WELLBEINGLWB, 2021-08-18:
Hi guys, when I follow the FCI documentation to use franka_gazebo, I got the following errors:

[ERROR] [1629248319.283261823]: Could not load class franka_example_controllers/CartesianImpedanceExampleController: MultiLibraryClassLoader: Could not create object of class type franka_example_controllers::CartesianImpedanceExampleController as no factory exists for it. Make sure that the library exists and was explicitly loaded through MultiLibraryClassLoader::loadLibrary()
[ERROR] [1629248319.283284655]: Could not load controller 'cartesian_impedance_example_controller' because controller type 'franka_example_controllers/CartesianImpedanceExampleController' does not exist.

[ERROR] [1629248320.284771, 0.208000]: Failed to load cartesian_impedance_example_controller

By the way, my OS is Ubuntu 16.04, and the installed versions of libfranka and franka_ros are 0.8.0. Could anyone help me get rid of this trouble? Many thanks in advance for your time and help!

gollth, 2022-07-05:
Hi @WELLBEINGLWB,

I just ran into your problem as well on Melodic. Took me some time to figure out why exactly this happened. In the end it works for me with using GCC but not using Clang++. Maybe try to switch back to GCC if possible and see if this helps. 

No idea why this can even be an issue. Must be something related with pluginlib but idk...

Hope this helps =)