frankarobotics/franka_ros #115: Are ther available implementations using franka_hw::FrankaHW and franka::Robot usable for the following scenarios

a-z-e-r-i-l-a, 2020-08-31:
I have a code for controlling Franka Emika robot that uses the [franka_hw::FrankaHW::control](http://docs.ros.org/kinetic/api/franka_hw/html/classfranka__hw_1_1FrankaHW.html#a75bd1438ffbfa54ce3ff6a4364df1cc0) method. This method takes a lambda function and calls ros_control controller manager update internally and has resulted in some limitations. 

Namely, while this control method is running, I can't call [franka::Robot::readOnce(](https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#a7ea7074a07b63fcf6933e97b078c7168))	to read the state of the robot and send it to another node for example. 

Secondly, if for some reason I want to manipulate the feedback to the controller, or the output from the controller, I can't do so (assuming I can't access the controller's implementation and can only access the ros_control hardware interface.

If I have understood correctly the above method internally uses [franka::Robot::control](https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#a5b5ba0a4f2bfd20be963b05622e629e1)  method. 

Do you know of any implementation that explicitely uses [franka::Robot::control](https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#a5b5ba0a4f2bfd20be963b05622e629e1)  method to make the controll loop ? 

Can you think of workarounds for the above scenarios if one had to still use [franka_hw::FrankaHW::control](http://docs.ros.org/kinetic/api/franka_hw/html/classfranka__hw_1_1FrankaHW.html#a75bd1438ffbfa54ce3ff6a4364df1cc0) method ?

gollth, 2022-01-05:
Hello @a-z-e-r-i-l-a and sorry for the late reply,

>  This method takes a lambda function and calls ros_control controller manager update internally

I'm not quite sure what you mean here. AFAICS the `FrankaHW::control()` method does not call any update of a controller manager:

https://github.com/frankaemika/franka_ros/blob/bd2bfd85eccd2be0c9e482f417b6c2d56bf2f73f/franka_hw/src/franka_hw.cpp#L221-L243

Rather the controller manager is updated in the main loop in `franka_control`: https://github.com/frankaemika/franka_ros/blob/bd2bfd85eccd2be0c9e482f417b6c2d56bf2f73f/franka_control/src/franka_control_node.cpp#L124

---

> Namely, while this control method is running, I can't call franka::Robot::readOnce() to read the state of the robot and send it to another node for example.

This is expected behaviour since with ROS control the `franka_control_node` is calling [Robot::control()](https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#a5d39a89a41edee89a776c3491dd20738) (of libfranka) which does the spinning for you. As mentioned in the docs of [Robot::readOnce()](https://frankaemika.github.io/libfranka/classfranka_1_1Robot.html#a7ea7074a07b63fcf6933e97b078c7168): 

> Cannot be executed while a control or motion generator loop is running.

---

> Secondly, if for some reason I want to manipulate the feedback to the controller, or the output from the controller, I can't do so (assuming I can't access the controller's implementation and can only access the ros_control hardware interface.

This is more tricky. AFAIK this is not easily possible in ROS control right now. There is [this discussion](https://github.com/ros-controls/ros_control/issues/289) where some people claim to have implemented "nested controller managers". Can this be an option for you? Create a custom ROS-control controller as a wrapper around your controller which adjusts the states & commands to your liking.

---
> Do you know of any implementation that explicitely uses franka::Robot::control method to make the controll loop ?

Not to my knowledge sorry

gollth, 2022-01-05:
Generally I'd say the approach would be to not fight but rather make use of the ROS control framework. Reading the state of the robot and sending commands to it, is can be done by writing your own ROS controller and claiming some robot interfaces.

Maybe a bit of context what you are trying to achieve here would also help to answer your question

gollth, 2022-02-03:
I'll close this now because of inactivity, but feel free to reopen, should your problem persist

closed 2022-02-03 by gollth