frankarobotics/franka_ros #117: ROS 2 support

mahaarbo, 2020-09-19:
I am currently trying to port the `franka_ros` package to ROS 2, and have managed to get `franka_msgs` and `franka_description`, and most of `franka_gripper` so far.  
Are there any plans for ROS 2 support? Is anyone else working on this as well?

gavanderhoorn, 2020-09-19:
This driver is heavily dependent on `ros2_control`. That's in-progress, but not finished yet.

Without that, I believe it's going to be difficult to get things to work in ROS 2.

mahaarbo, 2020-09-19:
Yup, that's where things stopped for me. But `ros2_control` is starting to take shape, so I think development alongside it can be interesting to do. If there is little to be discussed on this point now, I can close the issue.

gavanderhoorn, 2020-09-21:
> If there is little to be discussed on this point now, I can close the issue.

Just a clarification (if needed): I'm not a maintainer nor a dev here. I just wanted to point out what I believe to be the main blocker right now.

As to whether a ROS 2 port makes sense or is desirable: that would be up to Franka Emika (@fwalch et al.)

DLu, 2021-03-10:
`ros2_control` seems to be in better shape. I move that we re-open this.

mahaarbo, 2021-03-11:
As far as I can tell, when switching modes in the controller (e.g. going from joint-position level control to torque control), we must switch which callback function is bound to `libfranka`'s `robot` object ([franka_hw.cpp#L399](https://github.com/frankaemika/franka_ros/blob/kinetic-devel/franka_hw/src/franka_hw.cpp#L399)). Currently `ros2_control` does not notify the hardware interface about which command resource has been claimed (See: [ros2_control_demos#59](https://github.com/ros-controls/ros2_control_demos/issues/59), [ros2_control_demos#61](https://github.com/ros-controls/ros2_control_demos/pull/61)). 
How to best notify the `hardware_interface` that a new command resource has been claimed by a controller, and preparing for and switching is currently being developed and discussed at  [ros-controls/roadmap#41](https://github.com/ros-controls/roadmap/pull/41), and [ros-controls/roadmap#42](https://github.com/ros-controls/roadmap/pull/42). When this is finalized it should be possible to move to `ros2_control`.

mahaarbo, 2021-06-30:
I've started a [port](https://github.com/mahaarbo/franka_ros/tree/foxy-devel) of `franka_ros` to ROS2 foxy.
Since the structure of the controller manager is different this time around, it works similarly to the `FrankaCombinableHW` with the `robot.control` loop being set up in a separate thread. The implementation works, but since the `ros2_control` thread and the `robot.control` threads are not synchronized the motion is not smooth. 
Any help would be appreciated!

sabaghian, 2021-08-24:
Thank you for your input. We also think that ROS2 is starting to take shape. We have started looking into it and will announce in community as soon as the first release is ready.

marcbone, 2021-12-14:
Christmas is early this year. Head over to [https://github.com/frankaemika/franka_ros2 ](https://github.com/frankaemika/franka_ros2) and try out the first beta version of franka_ros2!

closed 2021-12-14 by marcbone