frankarobotics/franka_ros #118: franka_control.launch result in end-effector disconnected

Jekyll1021, 2020-10-15:
If I run gripper grasp app from DESK, the task is completed properly. However, after I run `roslaunch franka_control franka_control.launch load_gripper:=true robot_ip:=172.16.0.2`, DESK showed "end-effector disconnected" in Robot Status, and any request through franka_gripper/move or franka_gripper/grasp failed as a result. 

I got no error in roslaunch. Where should I start debugging this problem?

sgabl, 2020-10-21:
Hi, during the time you're connected to the FCI it's fine that DESK shows "end-effector not connected". However, the gripper move should not fail when the correct parameter are used. You could test if [the libfranka grasp example](https://github.com/frankaemika/libfranka/blob/master/examples/grasp_object.cpp) works.

bibbygoodwin, 2021-05-27:
I'm also having this problem - with interface v4.0.4 and the latest libraries: libfranka 0.8.0 and franka_ros 0.7.1.

As soon as I run `roslaunch franka_control franka_control.launch`, the web interface says "End effector not connected". When subsequently trying to send commands through franka_ros, for example with `panda_moveit_config`, planning is unsuccessful in this state.

MadBot97, 2021-08-10:
Hi, I had the same issue. Have you been able to manage it? Thanks.

matteolucchi, 2021-09-28:
Hi, I am also facing this problem with libfranka 0.8.0 and franka_ros 0.7.1 . Does anyone have any leads on how to solve this?

MadBot97, 2021-09-28:
Hi @matteolucchi, at the end I solved the problem by installing libfranka 0.7.1 and franka_ros 0.7.0. Be sure when you unblock joints from franka desk that also the gripper moves (otherwise you need to activate the gripper connection in the settings of franka desk). Hope this can help you!

gollth, 2022-01-03:
Hi all,

to amend a bit to @sgabl 's answer:

> "End effector not connected"

This warning is "expected behavior" since you can only control the robot *either* via FCI *or* via Desk. See the "switch" before the "robot control" block in [this graphic](https://github.com/frankaemika/the_reference_platform/blob/main/fig/overview_system_architecture.png) for more details. Regardless of the warning in Desk, it should be possible to control the gripper via ROS actions or libfranka commands (like it seems to be the case for you @Jekyll1021).

That action calls return errors can have different reasons. The most likely one is, that your goal width and tolerances are too narrow. Note that when using `franka_gripper/grasp` (or `franka_gripper/gripper_command` for that matter) the fingers will _always_ close (or open) fully until they feel resistance. Only then the current finger width is checked against the one you provided as goal. If it lies within the lower & upper bounds you provided, the action succeeds. Refer also to [the documentation](https://frankaemika.github.io/docs/franka_ros.html#franka-gripper) about this.

Lastly the gripper must be able to know where its hard limits are, which it does by homing its fingers inside the carriage. This is usually done when you unlock the brakes. Alternatively you can call the `franka_gripper/homing` action manually. This implies that you have setup a Franka Hand in the Settings Panel of Desk:

[image]

Let us know if this helps @Jekyll1021

maxiuw, 2022-09-17:
I followed the last step posted by  @gollthm. It did work however, main problem seem to be the gripper being 'open too much'. Fixed by manually closing the gripper and applying 'homing'

closed 2022-09-19 by gollth