frankarobotics/franka_ros #119: Release 0.7.0 for ROS Melodic

simonbogh, 2020-10-19:
Hi,

Will release 0.7.0 be pushed to ROS deb? Currently, I only get 0.6.0 on Melodic and e.g. `safety_distance` in the `franka_description` is not available and you get the following error when trying to instantiate the dual-arm example or load the robot arm to your own simulation with the new parameters.

    Invalid parameter "safety_distance"

sgabl, 2020-10-22:
Hi, just released `0.7.1` and created a [PR](https://github.com/ros/rosdistro/pull/27097) - packets should be available in a few weeks.

simonbogh, 2020-10-22:
Thanks. Will there be a release for Noetic as well?

sgabl, 2020-10-22:
As soon as https://github.com/ros-planning/panda_moveit_config/issues/72 is done, we will release `franka_ros` fully for ROS Noetic.

YijiongLin, 2021-01-12:
Hi @sgabl ,

Could you please take a look at the comment I posted [here](https://github.com/ros-planning/panda_moveit_config/issues/72)?

Any idea would be highly appreciated!

closed 2021-01-28 by simonbogh