frankarobotics/franka_ros #120: franka_ros controllers crashing

QingkaiLu, 2020-12-23:
1. I am able to 

- ping the control box ( `sudo ping 172.16.0.2 -i 0.001 -D -c 10000 -s 1200`)
[image]

- run the communication_test example of libfranka with one warning (`./communication_test 172.16.0.2`)
[image]

- show the robot state using the echo_robot_state example of libfranka (`./echo_robot_state 172.16.0.2`)
[image]

- show the robot in rviz using franka_visualization of franka_ros (`roslaunch franka_visualization franka_visualization.launch robot_ip:=172.16.0.2 load_gripper:=true`)
[image]

2. But when I run franka_ros controller examples, it crashes with the warning and errors below:
    > [WARN] [1608682350.969372]: Controller Spawner couldn't find the expected controller_manager ROS interface.
    [ERROR] [1608682380.427056883]: FrankaHW: Failed to initialize libfranka robot. libfranka: Connection timeout
    [ERROR] [1608682380.431456846]: franka_control_node: Failed to initialize FrankaHW class. Shutting down!
    
    Please see these five images below for the detailed outputs of running the franka_ros position controller example( `roslaunch franka_example_controllers joint_position_example_controller.launch robot_ip:=171.16.0.2 load_gripper:=true`):

    [image]
    [image]
    [image]
    [image]
    [image]

3. System and software information:
    **panda system:** 4.0.4
    [image]

    **panda FCI** is installed
    [image]

    **linux real time kernel:** 
    
    > Linux baiduusit-ThinkPad-P50 5.4.10-rt4 #1 SMP PREEMPT_RT Fri Dec 18 17:06:56 PST 2020 x86_64 x86_64 x86_64 GNU/Linux
    
    **libfranka:** 0.8.0 installed from source
    **franka_ros:** melodic-devel branch installed from source
    **ros:** melodic

zhengshuai1, 2021-01-04:
hi, as NVIDIA binary drivers are not supported on PREEMPT_RT kernels, do you know how to make drivers compatible with real-time kernels?

QingkaiLu, 2021-01-04:
> hi, as NVIDIA binary drivers are not supported on PREEMPT_RT kernels, do you know how to make drivers compatible with real-time kernels?

You can use one real time kernel machine without Nvidia drivers for control and use another machine with Nvidia drivers for perception. You can connect the control and perception computer using ROS. BTW, you should create a separate git issue for your question since I don't think the it is related with my issue.

zhengshuai1, 2021-01-05:
> > hi, as NVIDIA binary drivers are not supported on PREEMPT_RT kernels, do you know how to make drivers compatible with real-time kernels?
> 
> You can use one real time kernel machine without Nvidia drivers for control and use another machine with Nvidia drivers for perception. You can connect the control and perception computer using ROS. BTW, you should create a separate git issue for your question since I don't think it is related with my issue.

thanks, I have met the same controllers issue as you.

mahaarbo, 2021-01-25:
When you are running the visualization you are connecting to the IP `172.16.0.2`, but when you're running the joint position example controller you seem to be connecting to `171.16.0.2` was this the issue? That would explain the timeout.

QingkaiLu, 2021-01-27:
> When you are running the visualization you are connecting to the IP `172.16.0.2`, but when you're running the joint position example controller you seem to be connecting to `171.16.0.2` was this the issue? That would explain the timeout.

That makes sense. That is a dumb mistake, sorry about that. I will test it on the robot again once I can work onsite.

QingkaiLu, 2021-03-03:
> That makes sense. That is a dumb mistake, sorry about that. I will test it on the robot again once I can work onsite.

I tested the joint position controller example with the correct IP and it works now. Thank you for the help!

closed 2021-03-03 by QingkaiLu